hzeller/rpi-rgb-led-matrix #985: --led-show-refresh unreliable or --led-row-addr-type=3 slower than --led-row-addr-type=0?

marcmerlin, 2020-02-14:
So, I'm a bit confused with the numbers I see from --led-show-refresh
The instant number is often fairly different from the average number that gets printed when I exit with ^C. 

That said, I was benchmarking 128x64 panels in ABCDE vs ABC (the support you added in https://github.com/hzeller/rpi-rgb-led-matrix/issues/823).
My understanding was that ABC should be marginally faster since you only have to cycle through 8 combinations of lines to load, and the lines will be a bit brighter than ABCDE where you're loading 64 lines that don't stay on as long.
In other words, ABC should be superior to ABCDE both in brightness and speed, at least by a bit.

But my tests show otherwise:
```
root@rPi3:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-chain=1 --led-show-refresh --led-parallel=3   --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=30 --led-pwm-bits=7 --led-panel-type=FM6126A  --led-row-addr-type=3 -D0 
```
Gives an instant number of about 360Hz and 298.7Hz when I exit (^C)

```
root@rPi3:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-chain=1 --led-show-refresh --led-parallel=3   --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=30 --led-pwm-bits=7 --led-panel-type=FM6126A  --led-row-addr-type=0 -D0 
```
gives an instant number of about 450Hz and 406.3Hz when I exit.

Putting aside the Hz values that are off by 50 or so between instant and aggregate, why is ABCDE faster than ABC?

rowanG077, 2020-02-14:
ABCDE should be faster. ABCDE based panels select the line by outputting a 5 bit number at once. ABC panels clock in that same 5-bit number serially. And not even efficient at that. It requires 64 rising edges clock making a line switch significantly slower about ~192 times as slow.

hzeller, 2020-02-14:
if the multiplexing is the same, the ABC panels will not change anything. It is just a different way to get the row address into the panel.

Only having ABC lines can mean different things. You confuse it with panels that use lower multiplexing, so a 32 high display that uses 1:8 multiplexing (using ABC) vs. 1:16 multiplexing (ABCD). The former are typical 'outdoor' panels. What you have is a display that uses three lines to clock in the 5 bit number used for the 1:32 multiplexing.

Litmus test: If you have to set `--led-row-address-type`, then you ABC just means a different way to clock in stuff and won't make anything faster (if anything, a little bit slower). If you have to set `--led-multiplexing`, then ABC means you only do 1:8 multiplexing and that will be faster. I have only seen these for 32x32 outdoor panels or 32x16 (AB) outdoor panels. All the 64 panels high things I have seen so far do a variant of 1:32 multiplexing.

You will be slightly slower with the ABC in this case as it takes now at least 5 cycles to clock in a new row address vs. instant all bits on.

You might see a different last shown frequency because you just pressed Ctrl-C and there was some processing time lost due to that, so the last frame it displayed took longer.

marcmerlin, 2020-02-14:
OMG, thanks @hzeller , sorry that I was being dense and just didn't understand that someone somewhere had the briliant ID that saving writing 2 pins on the connector, but still passing 5 bits over 3 pins, was a grand idea somehow.
Now it totally makes sense that ABC would be even worse than ABCDE. I naively thought they made a step in the right direction and shifted 512 bits over each address line, making the panel a /8
Ok, now I understand that I should avoid ABC panels, unless they are true /8 panels, which from what you've seen and lack of reports that they exist, don't actually exist at the moment.

The ^C issue is pretty consistent and makes it hard to benchmark panels. It would be cool if it updated an average Hz counter and printed that when you exited. Yes, PRs welcome, I know :)

marcmerlin, 2020-02-14:
Alhought the hertz display counter issue isn't fixed per se, the main point of this bug, was not a bug, but unfortunate expected behaviour, so I'll close this issue.

closed 2020-02-14 by marcmerlin