hzeller/rpi-rgb-led-matrix #986: P4 80x40 10S (1:10 scan) multiplexer help

SharukaV, 2020-02-17:
Hello.
I am trying to get a panel labeled P4 (1921) -80x40-10S-1.1 to work.

The image is duplicated in the case of a running line at any multiplexing.
Almost all the options for the demo options I tried.
Could you tell me the settings you need to explore?
Is it possible to solve the problem by writing a multiplexer?

Attached two photos, on the first square and three dots of different colors - duplication is visible
On the second, the running line - regardless of the multiplexer - the panel is divided into two from the seridine, the height changes but the duplication is constant.
[image]
[image]

master-bunny, 2020-02-17:
same here, qiangli p4 80x40 rgb matrix. i'm using "passive-3" adapter and has 4 blank lines at middle (after 16 rows) and at bottom (after 36 rows) any flags cant help. Hub75e of course

caxepok, 2020-12-03:
> same here, qiangli p4 80x40 rgb matrix. i'm using "passive-3" adapter and has 4 blank lines at middle (after 16 rows) and at bottom (after 36 rows) any flags cant help. Hub75e of course. this 4 blank lines is appeared at top of starting block (17-20 is drawing over 1-4 lines, and 37-40 is on top of 21-25)
> [image]

Hi. Have you solve your problem?

hzeller, 2020-12-03:
What parameters did you give to the binary, in particular the `--led-cols` and `--led-rows` flags?

muharramapri, 2020-12-13:
I also have the same problem with the same panel, until now the problem has not been resolved. Can it help me?
I used --led-cols = 64 --led-rows = 64
[image]

muharramapri, 2020-12-13:
> What parameters did you give to the binary, in particular the `--led-cols` and `--led-rows` flags?
I also have the same problem with the same panel, until now the problem has not been resolved. Can it help me?
I used --led-cols = 64 --led-rows = 64
[image]

hzeller, 2020-12-13:
But if it is an 80x40 panel, wouldn't you want to use `--led-cols = 80 --led-rows = 40` ?

muharramapri, 2020-12-13:
> But if it is an 80x40 panel, wouldn't you want to use `--led-cols = 80 --led-rows = 40` ?

I've tried it but it's still like that

peragwin, 2021-08-28:
has anyone discovered the correct multiplexing to use for this panel?

cPabz, 2021-12-23:
So these screens are outdoor screens and maybe internally connected as a chain of [ 16(columns) x 40(rows) with led chain = 5. ]
I have the same outdoor board, and successfully got one screen to work with the following command flags
`--led-cols=16  --led-rows=40 --led-chain=5 --led-multiplexing=1 --led-row-addr-type=3`

If you are using chained screens as explained in the ReadMe, you can link more screens with the same chain.

But I also have a doubt, this is not a perfect solution, we need to have a multiplex mapper for our board, or else when we are using multiple boards in different use cases, we might need the pixel-mapper flag, which if used will not work as excepted.

sbsiaf, 2022-06-04:
Hi, recently I had a problem trying to configure the qiangli P4 80x40 panel using the 2038s IC. Configuration is rather simple. I could not use the HD2020 to configure, but could do it from the LEDART android app. You have to create a hardware profile in the app and then save it and manually upload the same to the controller.  Thereafter you can use any computer software to send your DATA. The steps are as follows:
1.  On a mobile running LEDATR, in the device options, hardware menu, go to smart settings.
2. select width 80 and IC 2038S. (in HD2020 application on computer, the option is max of 64 which gives an error on the module)
3. from the next screen, just follow the directions.
4. You will see two lines on the controller. After you select 2, in the no of lines option, one LED will blink on the top left in the next screen.
5. in the next screen, you have to select and match the matrix of the blinking LED on P4, to that of the next screen on mobile.
6. After all LEDs match, you will be asked to give a name to this new hardware profile and save.
7. Now select this as the device in device options and your display will be ok.
Note, pl do  this config on a single display, and with initial screen selected to 80x40. After configuration, you can increase the width and height depending on the number of panels used.

athulvingt, 2022-07-13:
> Hi. Have you solve your problem?

Hi Have you solve your problem?

b4rtek, 2022-10-28:
Hi, i try to connect p4 80x40 hyb75 led matrix to my raspberry pi b+ via idc cable ... but its not working.
I use manual from https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md 
I try this settings --led-cols=16 --led-rows=40 --led-chain=5 --led-multiplexing=1 --led-row-addr-type=3 - no results :(
In manual pin 10 to E  must by connected to ground below g2 on idc cable? 
Please help :)