hzeller/rpi-rgb-led-matrix #987: Python image-viewer.py example code

ElBertrando, 2020-02-18:
Hi,
I am using a pi 4 with the adafruit HAT with the pwm. I have a 32 x 64 panel that I am simply trying to display an image to. 
I see that the image-viewer.py code requires and argument with the line "image_file=sys.argv[1]" , so I created a jpg image that is 32 x 64 pixels [the size of my rgb scree] called "Recording.jpg" and opened up terminal and ran “sudo python ./image-viewer.py Recording.jpg –led-gpio-mapping=adafruit-hat-pwm –led-rows=32 –led-cols=64 –led-slowdown-pgio=4” ---
However now the image is only taking up the top third or so of the panel and is looking horrible. I am trying to figure that out now, because I created the image in Photoshop to have the exact amount of pixels or the panel.
I am able to have it fill the panel by editing the image-viewer.py code "option.cols = 64", but the image is still horrible and flickering -- Any help will be appreciated.
Thanks,
Bret

hzeller, 2020-02-18:
Did you try first with the native image viewer in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#image-viewer ? This would be the recommended way to display an image.

I think the Python code uses some function to resize the image, maybe that gets tripped with your image size. Have a look at the code.

(Also, if you want to have an image the right size, I'd try to store the image as PNG image to avoid jpg artifacts which will be quite pronounced with these few pixels)

ElBertrando, 2020-02-18:
Hi -- I am now testing with the "lcd-image-viewer.cc" code in the Utils direcotry.
I created two .png's  -- one with just a red background and one with white text on a red background that says "recording". The red background displays fine of course but the one with the white text looks unreadable. So now I am thinking I might be doing this wrong. 
If I want to display a red background with white text which says "recording" -- is sending an image file the best way ? I want to just be able to send white text with a red background directly to the panel for a studio environment... 
Thanks,
Bret

hzeller, 2020-02-18:
In what sense is it unreadable ? Is the text mumbled up in some way or is it just the contrast ?

If the text does not show right, test out if you have one of the other multiplexing types in your matrix (test with [`--led-multiplexing`](https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing))

If it is the contrast, it is often a good idea to put a darker one-pixel outline around each letter to show up with more contrast.

ElBertrando, 2020-02-19:
Here is the picture I am trying to upload https://imgur.com/3zqT3YH The image size is 64x32.  However the white letters are all blended together and seem to be flickering. The --led-multiplexing commands do not work unfortunately.  I took a pic with my phone of what I am seeing: https://imgur.com/v0aifN0   Is there a recommended font t use in Photoshop ? I have been using Arial -- 
Still working on this -- Actually I am starting to look at the runtext.py program to post a static text with a different colored background as an alternative

hzeller, 2020-02-19:
What Raspberry Pi are you using ? If it is one of the newer ones, they might be too fast for the panel. In that case, try to [`--led-gpio-slowdown`](https://github.com/hzeller/rpi-rgb-led-matrix#gpio-speed)

Can you try the demo first to see how it looks on your panel ? [`examples-api-use/demo`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#running-some-demos)

ElBertrando, 2020-02-19:
Pi4 . Yes I need to use --led-gpio-slowdown=4.  
I found if I run this command “sudo ./text-example --led-cols=64 --led-rows=32 -f ../fonts/7x13.bdf --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4 -B 252,2,2 -C 255,255,255” I can get a red background with white text, but the background only is behind the text and not the entire panel -- Is there a parameter to set the background for the entire panel ?

hzeller, 2020-02-19:
There is no option, but you can modify this example program to do what you need:

```patch
--- a/examples-api-use/text-example.cc
+++ b/examples-api-use/text-example.cc
@@ -146,13 +146,14 @@ int main(int argc, char *argv[]) {
            "Supports UTF-8. CTRL-D for exit.\n");
   }
 
+  canvas->Fill(bg_color.r, bg_color.g, bg_color.b);
   char line[1024];
   while (fgets(line, sizeof(line), stdin)) {
     const size_t last = strlen(line);
     if (last > 0) line[last - 1] = '\0';  // remove newline.
     bool line_empty = strlen(line) == 0;
     if ((y + font.height() > canvas->height()) || line_empty) {
-      canvas->Clear();
+      canvas->Fill(bg_color.r, bg_color.g, bg_color.b);
       y = y_orig;
     }
     if (line_empty)
```

In general, it is probably advisable to just write your own version of this program. If all you need is showing a static string, this can be much easier of course.

hzeller, 2020-02-19:
In https://github.com/hzeller/rpi-rgb-led-matrix/commit/051af97e732248e47f507883785de393f2de9418  I've now added a color flooding parameter `-F` which allows to choose the panel background color.

ElBertrando, 2020-02-19:
I'll download the latest version and see what I can do -- I will report back -- I appreciate all of your help today

hzeller, 2020-02-19:
I've now also adapted the scrolling-text-example to accept a flood-color parameter and also allows to scroll with zero speed. This might be more useful if you just want to display a text without having to enter it manually on a line.

(Need to download even more latest version for that).

This example also shows how the 'outline' option allows us to create a slightly darker red outline around the text (127,0,0) to have the contrast of the white text punch out more nicely:

  * `-s0` zero scrolling
  * `-y5` shift the text so that it is somewhat in the middle of a 32x64 panel
  * `--led-cols`, `--led-rows`: as expected
 * `-f`  use a font with relative path in this distribution
 * `-B` background color of text
 * `-C` color of text
 * `-O` color of outline
 * `-F` flooding color of panel not text
 * `"Recording"` the text to display

```
sudo ./scrolling-text-example -s0 -y7 --led-cols=64 --led-rows=32 -f ../fonts/7x14.bdf -B255,0,0 -C255,255,255 -O127,0,0 -F255,0,0 "Recording"
```

ElBertrando, 2020-02-19:
Hi -- I have downloaded and replaced the "rpi-rgb-led-matrix" folder in my pi 4 with the latest "rpi-rgb-matrix-master".  I am running the command in terminal that you gave as en example above, but am getting the error: **/scrolling-text-example: invalid option --'F'**
However I have cracked open the code and can see the `case 'F': ` that you added in the while loop so I am not sure why I am getting this  error message..

hzeller, 2020-02-19:
did you recompile everything ? `make clean ; make` ?

ElBertrando, 2020-02-19:
Yes of course!!!  I had to revisit that again as its been a while since using c++ ( I am using python for this project).  I had to add a few variables but this works great: 
`sudo ./scrolling-text-example -s0 -y7 --led-cols=64 --led-rows=32 -f ../fonts/7x14.bdf -B255,0,0 -C255,255,255 -O127,0,0 -F255,0,0 "Recording" --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4`
I assume I can call this command from within my python code so I can change the text based on the event handlers of the recording device -- for example "RECORDING" to "PAUSED" -- I will be working on this now....

hzeller, 2020-02-19:
There are also text examples in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python/samples if you want to do this in Python.

ElBertrando, 2020-02-20:
I see that however it looks like the parameters are different for the "runtext.py" sample than the "rolling-text-example.cc" file. Do I have all of the same options available [-O, -B, -C,-F], as well as the -s0 option ?

hzeller, 2020-02-20:
you would need to implement these. But the features to set the color and background color etc. are also available in Python, you just need to do the equivalent things as in the C++ code.

ElBertrando, 2020-02-20:
I think you are talking about changing the core.cpp [or maybe the core.pyx] file in  **\rpi-rgb-led-matrix-master\rpi-rgb-led-matrix-master\bindings\python** to add the functionality ?

hzeller, 2020-02-20:
All you need is graphics.DrawText() and canvas.Fill(), both of which are already provided in Python.  From that, you can implement the functionality with the various options.

You have to check if the font function that generates an outline font is already there.

ElBertrando, 2020-02-20:
Hi -- I see that runtext.py inherits from samplebase, which in turn imports RGBMatrix and RGBMatrixOptions from "core.pyx" ? 
It looks like I can use runtext.py as a model to obtain graphics.DrawText() and canvas.Fill(). 
I can then pass in arguments ("text", "--led-rows" etc) into the fucntion like this `runtext.Run(text="hello", --led-rows=32,etc...) `

ElBertrando, 2020-02-26:
Hi Henner,
I have been taking a different approach to run the script from my file with **subprocess.call()** - from my main program I am using     `PassT.subProcScrollingRECTest("-s0","-f ../fonts/7x14.bdf","-B255,0,0","-C255,255,255","-F255,0,0", "-O127,0,0")`

which is calling:
```
def subProcScrollingRECTest(scrollRate, tFont, bColor, cColor, fColor, oColor):
    subprocess.call(["/home/pi/rpi-rgb-led-matrix-master/examples-api-use/scrolling-text-example", "-y7", "--led-cols=64","--led-rows=32","--led-gpio-mapping=adafruit-hat-pwm","--led-slowdown-gpio=4",'"Recording"', scrollRate, tFont, bColor, cColor, fColor, oColor])
```

Getting much closer but still error code: **Couldn't load font '  ../fonts/7x14.bdf'**

hzeller, 2020-02-26:
The relative path of the ../fonts file is of course sensitive from where you start it. So I'd use the full absolute path to the font-file.

ElBertrando, 2020-02-27:
Hi Henner,
Yes this works of course. My next problem is to kill the subprocess programatically. Currently I am using subprocess.Popen to obtain the pid of the return. Then use the command `os.kill(x, signal.SIGKILL)` to end the program where "x" is the **return.pid**  -- 
However I am getting an error because I need to run as root. Does this mean I should once again use subprocess to run this command as root ? Am I on the right track ?

hzeller, 2020-02-27:
I still think the right track would be, if you want to use Python, to directly use the Python API and do things from there, not running subprocesses.

UCSF-ETS, 2020-03-17:
Hi Henner,
I pretty much have the subprocess.call() method pretty much finished. 
However I am deciding to take your advice and now reprogram it using the API. For the programming I am doing it from my windows machine in visual studio code and then SFTP over to the Pi4. 
This might be a stupid question but as far as using the API, how do I import the rgbmatrix module. I know i can't install with "pip install rgbmatrix", but when I put the "rgbmatrix" folder in the same folder as my program -- the line "from rgbmatrix import graphics" still brings up an error.

UCSF-ETS, 2020-03-18:
I have succeeded with the subprocess.call() technique. But I agree now I should use the API. However even though I am placing the "rgbmatrix" folder in the same directory as the rest of my .py code files, I am getting error importing it "from rgbmatrix import graphics" -- using the runtext.py code as my guide

ElBertrando, 2020-03-18:
I have succeeded with the subprocess.call() technique. But I agree now I should use the API. However even though I am placing the "rgbmatrix" folder in the same directory as the rest of my .py code files, I am getting error importing it "from rgbmatrix import graphics" -- using the runtext.py code as my guide

ElBertrando, 2020-03-18:
I am reading the python bindings instructions -- however this command is not working as well:
You can also build for Python 3:

```
sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)
```

I am running python 3.8

hzeller, 2020-03-18:
what do you mean with 'not working well' ? Do you have an error message to share ?

ElBertrando, 2020-03-18:
Perhaps I'm doing something wrong.. 

> sudo apt-get update && sudo apt-get install python3-dev 
python3-pillow -y
At line:1 char:21
+ sudo apt-get update && sudo apt-get install python3-dev python3-pillo ...
+                     ~~
The token '&&' is not a valid statement separator in this version.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : InvalidEndOfLine

hzeller, 2020-03-18:
This looks like you're using a funny shell.
You have to install this on the Pi, and the shell should be standard bash.

ElBertrando, 2020-03-18:
OK -- I am using visual studio code in windows 10 to write the code, then SFTP to transfer over to the pi4 --

ElBertrando, 2020-03-18:
Any way to make this work as far as referencing the library in the folder "rgbmatrix"  from visual studio code in windows ?

hzeller, 2020-03-18:
Don't know, probably not. Simplest is to use Linux on your workstation, there it should compile, even if it can't connect to the GPIO and matrix.

ElBertrando, 2020-03-18:
I'll give that a try --

UCSF-ETS, 2020-04-14:
Hi Henner,
I now have an ubuntu box running virtually on my machine and am working on using the RGBMatrix API technique. I am using the "runtext.py" example code for Python. I see the code inherits Samplbase, but has added the line 
`self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel", default="Hello world!")`
For controlling the scrolling text, I do not see the argument in samplebase or runtext.py. I suppose I need to add the line:
`self.parser.add_argument("-s", "--scrollingtext", help="The scroll rate", default="-s0")`  ?

ElBertrando, 2020-04-14:
Hi Henner,
I now have an ubuntu box running virtually on my machine and am working on using the RGBMatrix API technique. I am using the "runtext.py" example code for Python as a start. I see the code inherits Samplbase, but has added the line
self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel", default="Hello world!")
For controlling the scrolling text, I do not see the argument in samplebase or runtext.py. I suppose I need to add the line:
self.parser.add_argument("-s", "--scrollingtext", help="The scroll rate", default="-s0") ?

ElBertrando, 2020-04-15:
I guess I could simplify my above questions to ask if the python API provides a parameter to adjust the rate of scrolling text in the runtext.py the way the scrolling-text-example.cc does  ?

hzeller, 2020-04-15:
I don't have the code in front of me right now, but since you're editing it, you will certainly see if such a feature is implemented ?

ElBertrando, 2020-04-15:
I have been looking and don't see the same control for scroll rate in runtext.py as there is no reference for "speed" as there are in the c++ examples.

hzeller, 2020-04-15:
If you want to implement it, please send a pull request.

ElBertrando, 2020-04-15:
Pull request created --

ElBertrando, 2020-05-08:
Hi Henner,
It's been a while -- I was checking on pull request #1054 -- thx

ElBertrando, 2020-06-12:
Hi Henner,
I updated my pull request to include implementation -- Are we moving in the right direction ?

Jmorel06, 2022-01-07:
Hey, good afternoon. How can I change the size of the letters on my led display following this: https://github.com/hzeller/rpi-rgb-led-matrix.

I want to present them in a larger size if possible.

Thanks