hzeller/rpi-rgb-led-matrix #988: Question only: Enough speed on Pi (or Zero W) for simple display?

gingters, 2020-02-19:
Hi,

I have a question, and before I buy multiple LED panels just to try something out and then figure out it doesn't work, I think I'd rather ask the experts ;)

I want to display a pretty simple Text/Number in only 3 colors, like [AA 00:00] in either red, yellow or green (or maybe the clock part in white and the letters in one color, or vice versa). The clock-part needs to show a countdown, so it only updates once every second. A nice to have would be to be able to display a one-colored scrolling text on the display, like "Pause until 14:15" or "Welcome to our tournament".

First question for you experts who had these displays in their hands already:

That display should be readable outdoors from up to 20 or 30m away, so I guess its not enough to use only two 32x64 panels side-by side (32 x 128) for that text to be readable that far away but I probably need two rows of four panels each (64 px height x 256 px width in total)?
Or do you think a single row of two panels would be large enough to see the countdown in bright sunlight? (which I would prefer, because of costs ;) ).

I need to synchronize at least two, possibly multiple displays at different locations, so the idea is to have a server, and multiple displays each controlled by one Raspberry, connected to the others via WiFi. That's why I thought about the Raspberry Pi Zero W here as they are cheap.

The software to control the displays would be written in C# / .NET Core and ideally using .NET Native to compile directly to machine code (if possible), using this library.

So that is question two:

Do you think a Raspi Zero W is fast enough to communicate with a server via WiFi (SignalR / WebSockets) and then control the display accordingly? The scrolling text does not need to be synchronized, but the running countdown needs to be the same on every display within a few milliseconds. Or should I go for a Pi 4 here?

Thanks for your time!

wolfi-by, 2020-02-19:
Hi!

Dotnet Core is possible... Maye Blazor would be an easy option

Use a Rpi3 or better a Rpi4... And also use a wifi router for longer distances because I Board access point is too weak

Have fun

hzeller, 2020-02-19:
The most important thing for outdoors is actually to get bright panels. If you look for so-called outdoor panels, then this is a good first indicator. Make sure that they claim a 'multiplexing' with a divisor that is 1/4 of the number of lines. So if you get 32px high panels, it should say 1:8 multiplexing, if it is 16 px high, it should say 1:4 mutltiplexing (or, even better 1:2). This will guarantee that you get a visible display even in bright outdoor conditions (it also will make the panels less flicker as a nice side-effect).

(also nice side effect of the outdoor panels: they are usually better sealed at the front to avoid getting messed up in the rain).

Then: size. If you want to have it seen outside, you probably want somewhat larger panels but don't care too much about hi-resolution. A typical  size of P5 (5mm pixel distance) is probably too small if you want to read it (or you need a lot of panels, but the resolution would be somewhat overkill).

There are panels with P8 or P10 out there (in particular the outdoor panels), and that is what I would shoot for. If it is to be seen from a distance, you want a larger panel, but don't necessarily care about the higher resolution. In the library,  you select the right multiplexing with `--led-multiplexing` (might require just going through all of these settings to see which particular kind your panel implements).

Also, I have never seen 64px high panels for outdoor use: the LEDs are simply too small to push out a lot of light. So it is better if you go with a P10. Cheaper, brighter, faster - I guess these might be good criteria here :)

So if you use 8 panels with 32x16 P10 outdoor panels with 1:4 or even 1:2 multiplexing you  can have two parallel chains with 4 panels each, creating a display of 128 pixel x 32 pixel , which translates into a probably nicely visible 128cm x 32cm display. If you add another parallel row, you need 4 more panels and have a 128cm x 48cm panel.

 (this is a typical listing found on a random aliexpress search; https://www.aliexpress.com/item/32382300128.html - there is of course always a caveat that this could be a new panel type might require some tweaks in the library - the vendors constantly seem to change things around),

Given that this is not too high of a resolution, and that it is faster multiplexing and that you only will probably use it with `--led-pwm-bits=1` (8 colors total), even a Raspberry Pi Zero or Zero W should do it (they are usually not really recommended, but for this case it might be sufficient).

The Wifi on a Pi zero OTOH might not be very good for long distances. You might put everything together at home and it works, but if you then have to bridge multiple tens of meters you might find it unreliable 'in the field'. So it is worthwhile spending some time figuring out if the wifi situation will work fine.
Even if it sounds like a bit of a hassle, I'd actually try to wire them up via wire ethernet (which means you'd need a Pi with an ethernet jack).

For synchronization and sending image content to multiple panels connected via the network, I personally would use the [FlaschenTaschen protocol](https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md#protocols) which sends the image content in a single UDP packet, so you can rapidly send the same packet to all your connected displays (never tried multicasting, but that should be doable as well).
The FlaschenTaschen Server can be [compiled to run with the RGB matrix](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display). There are [a few ready-made clients](https://github.com/hzeller/flaschen-taschen/tree/master/client#flaschen-taschen-clients) that allow to send text or images (or VLC has a native FlaschenTaschen mode to send videos...). Since it is a simple UDP packet that is easily supported by all languages, you can write your clients in C# if you like (right now, there is a C++ and Python API, feel free to send a pull request to https://github.com/hzeller/flaschen-taschen/tree/master/api to provide a C# class).

In general (and in particular for the smaller Pi's with only one core), make sure to not run anything else on the Pi that is memory churn intensive as it will result in some visible flicker on the panel. So for the code on the Pi i'd strongly suggest to not use anything that does garbage collection (e.g. Python or C#), but use predictable, low-overhead code in C++. Client side code which  you run on your laptop of course can be written in anything.

Beersatron, 2020-02-19:
I have used p5 indoor panels with pi3's for an outdoor scoreboard/display. This is my first version https://github.com/hzeller/rpi-rgb-led-matrix/issues/631 and this is my second version https://github.com/hzeller/rpi-rgb-led-matrix/issues/689 .

I can't remember the panels I bought, but for Version3 of my Scoreboard (when I get the time) I think I will follow what @hzeller just described and change to p10 outdoor panels with better multiplexing. That way I might even be able to take pictures and videos using a phone's camera!

gingters, 2020-02-19:
Wow. Thank you a lot for that detailed answer!

For testing purposes I'm going to order 2 panels and check if I get this to work. If that works I can upscale that to 4 or even 8 panels in 2 rows if that is needed for visibility. I think a single row with 2 panels should be enough for indoor usage anyways.

I have a Pi Zero W and a normal Pi 3 B for testing purposes, so if the Zero with the single core isn't fast enough I can try the 3. I'll probably try .NET Core / C# first, and write code with as less allocations as possible to prevent long GC cycles. If that does not work as good as hoped for I'm afraid I have to brush up my C++ or use Rust or Go or something similar that compiles no native binaries.

Now... I need to wait for the panels to arrive ;)

closed 2020-03-27 by gingters

Worldplas, 2021-11-15:
Hello,
I am trying to run the library on the new Raspberry Pi Zero2 https://www.raspberrypi.com/news/new-raspberry-pi-zero-2-w-2/  using python binding. 
When i run the sample runtext.py, i have weired colors on the monitor and after the raspberry hangs with black screen.
Did you have the issue, or someone had already tried the new Raspberry Pi Zero2 ?
Thanks in advance.

Worldplas, 2021-11-15:
Thanks hzeller, waiting for your feedback then.

hzeller, 2021-11-16:
I've added some detection for the Zero W 2, but I am currently travelling, so can't test with a LED matrix.
If someone here with a Zero W2 (e.g. @Worldplas ?) could do a fresh `git pull` and try if the Matrix works with it now ?

Worldplas, 2021-11-17:
@hzeller thanks alot, i will try it and let you know

Worldplas, 2021-11-17:
@hzeller i confirm that it works fine on Zero W2, we have to use --led-slowdown-gpio=2 as for the Rpi4.
Thanks for the modficiation

leissler, 2024-04-27:
Do you think a PI Zero W2 would be able to drive a 128x128 screen (made from 4 64x64 panels), if it is the only thing it does? The picture is actually generated on another Pi and sent over network to the Zero2.

davemaster, 2024-04-28:
> Do you think a PI Zero W2 would be able to drive a 128x128 screen (made from 4 64x64 panels), if it is the only thing it does? The picture is actually generated on another Pi and sent over network to the Zero2.

The only way (the easiest) to know is trying... good luck