hzeller/rpi-rgb-led-matrix #989: Driving 6x64x64 Matrix's from a single Pi?

mikecann, 2020-02-20:
Hi @hzeller,

First thanks for this incredible library mate, you have done an excellent job here. I have already used it to great effect creating an LED matrix toy here: https://mikecann.co.uk/personal/led-matrix-toy/

Now I want to build an LED Matrix cube. So that means driving 6 of the same panels that I have purchased here: https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.840e4c4dtoL1i7&orderId=8010499598137002&productId=32650789975

Those are `p2.5 indoor smd2121 hub75 64x64 led` `1/32 Scan`.. 

So my questions are:

Can I drive all of them together from a single Pi? 
Can I just daisy chain them together then connect an Adafruit Bonnet to the Pi and expect it to work?
Would a Rasperry Pi 4 work or should I go with a 3 or something else?

I appreciate you probably dont want to play technical support to every maker on the planet but ill definately be giving you massive props in my writeup if I can get this thing built and working 😄

hzeller, 2020-02-20:
You can drive these with a single Pi but I'd not suggest to use a long chain of six of these panels: that will definitely result in noticeable flicker as the panels are not fast enough to clock with the needed speed.

Use three parallel chains (e.g. using the [active adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3), [a ard party one](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/) or build your own), and then put only two of the 64x64 in each chain. So don't use the Bonnet as it only allows for one chain.

I suggest a newer Pi, so a 3 or 4 are great, Pi zero is probably too slow for this.

mikecann, 2020-02-20:
Awesome, thanks for that! I really appreciate you taking the time to give me some advice on this.

I havent ever tried to make PCB before from designs such as you have listed on your [active adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) page so to be safe I decided to go with one of the third party ones for now.. fingers crossed it works.

Ill grab a Pi 3 or a 4 too.

Im just trying to calculate how much power the displays will need. The sellers page (https://www.aliexpress.com/snapshot/0.html?orderId=8010499598137002) says that they need a maximum of 1200 W/m2. 

For this project I have 6 x 160mm x 160mm = 153600 mm2 = 0.1536 m2 .. x 1200 w/m2 = 184 w

So im probably going to need about 200w power supply at 5v thats 40a .. so thats quite a bit bigger than anything I have lieing around here..

mikecann, 2020-03-17:
@hzeller @raspberry-pi-maker Hey guys. My Pi and panels have arrived and I have been playing with it trying to get it to work and it sort of does... see this video:

https://photos.app.goo.gl/QiAD42SKG2d57Psy5

I havent modified the Electrodragon HAT or the Pi in any way.

I used DietPi Buster Lite as described in this post: https://github.com/hzeller/rpi-rgb-led-matrix/issues/991

Im not sure exactly whats going on.. my current thinking it is something to do with the 1:32 multiplexing and the `E-line` as mentioned at the end of here: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

My Matrix panels are these here: https://www.aliexpress.com/item/32650789975.html?spm=a2g0s.9042311.0.0.17ea4c4dE7aC3h

Is this something you have seen before? If it is the E-address line that I need to change could you hint as to which ones I should be soldering? 

[image]

raspberry-pi-maker, 2020-03-18:
@mikecann 
I successfully tested 64X64 matrix with Electrodragon HAT.
Perhaps you can use it with one of the following pictures.
This is the same concept of @hzeller 's explanation.(https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3)

In my case, the second picture(P8-E, P4-GND) is the successful config.

[image]
[image]

mikecann, 2020-03-19:
@raspberry-pi-maker thats awesome. Ill give it a try, hopefully I dont break anything :)

mikecann, 2020-03-19:
@raspberry-pi-maker It worked! Awesome :) Thanks again.. Going to keep this issue open just incase I run into any other issues when trying to get 6 of them running but so far its looking good!

mikecann, 2020-03-30:
@raspberry-pi-maker @hzeller okay so after a lot of pain I have the cube built but im running into another issue see this video:

https://photos.app.goo.gl/jZ5scPPoYUvJQRbG8

You can ignore the flickering, only the camera sees that you cant see it with the eye. The problem is the half / half on the side and top panel. 

This is the command im running:

`sudo examples-api-use/demo -D11 --led-rows=64 --led-cols=64 --led-chain=2 --led-parallel=3 --led-slowdown-gpio=2`

I was worried that the panels were broken or something but when I changed which port on the HAT I was plugged into I saw it happening with different panels. 

So I think the issue is the two panels that are plugged into P3 seem to suffer from half of each panel is missing the blue part of the signal.. 

[image]

Anyone seen this before? Could it be something to do with me plugging in the wrong e-line?

mikecann, 2020-03-30:
Hahaha typical... I found the issue! It was this switch:

[image]

I flicked it the other way and suddenly everything looks awesome :D

mikecann, 2020-05-20:
So finally I have finished this project :) I made a video that went through the whole process here:

https://www.youtube.com/watch?v=l_GSGFhnWwo&feature=youtu.be

Massive thanks to you @hzeller  and @raspberry-pi-maker for all your help with this one :)

closed 2020-05-20 by mikecann

daveythacher, 2020-10-12:
> Im just trying to calculate how much power the displays will need. The sellers page (https://www.aliexpress.com/snapshot/0.html?orderId=8010499598137002) says that they need a maximum of 1200 W/m2.
> 
> For this project I have 6 x 160mm x 160mm = 153600 mm2 = 0.1536 m2 .. x 1200 w/m2 = 184 w
> 
> Is my math right there?
> 
> So im probably going to need about 200w power supply at 5v thats 40a .. so thats quite a bit bigger than anything I have lieing around here..

It would probably be closer to:
64 * (64 / 32) * 6 * 45 = 3.84A * 5V = 172.8W
or
64 * (64 / 32) * 6 * 60 = 46.08A * 5V = 230.4W

Formula:
Columns * row scan * panels * current for RGB pixel (3 LEDs) * 5V

This is peak power. Without changing the current gain, if supported, PWM/brightness will only change average current. Peak power will set range for average current. You solve power problems with a huge power supply, tons of caps (may create a ton of problems), software current gain LED panels, multiple power supplies, etc.

daveythacher, 2020-10-12:
One easy way to limit peak power is to only turn on so many LEDs in a single row. Usually there is two active rows. For this setup there is a total of 768 RGB LEDs on at anytime. 

So If you turn on 75 percent of the LEDs you will see 75 percent of the Peak power as the new Peak power. This and the current gain directly controls the load which creates the peak power. PWM and caps do averaging tricks. While power supplies provide max peak power on standby.

mikecann, 2020-10-12:
@daveythacher hmmm.. In the end im pretty sure I wasnt supplying anything close to this calculated power requirements and it still worked.. see my video on this project where I talk about it here: https://youtu.be/l_GSGFhnWwo?t=243

So *shrug* I have no idea what is going on

daveythacher, 2020-10-12:
On battery that peak current would have mattered a lot. Hopefully there was a discharge protection circuit in there. Even with 4 6+Ah battery packs you could easily be pulling too much power. Assuming 1C discharge per battery pack you would have 26A, but could attempt to draw 34.56A to 46.08A. 

Adafruit 6.6A packs only work at 0.5C which means 13.2A. Note Adafruit sells 6.6Ah battery packs but the wire is only rated for 2A. So in reality you would only have 8A.

Again if you limit the number of LEDs that are on in any given row you could be under that limit. In this case it will work and not have an issue to my knowledge.

There is another possible problem which is you appear to be using a buck booster inside your charge circuit. To step up the battery voltage to 5V. These have a current limit of 2A each so you are still at your 8A limit to the panels. However the way you convert a lower voltage to a higher voltage is by increasing the current. Note this process is not completely efficient. 

This means you are getting 8A on the panels, but are pulling more current from battery. Say about ((5/3.7)/.85)*8A=12.7A. This would put you at about 0.5C on the batteries. However exceeding the battery wire/connector limits by 1.175A. The converter will likely shutdown if you try to exceed 8A to the panels, but it may start happing at a lower point.

It is probably pretty close. However I would avoid taking it to its limit. I would also avoid using a ton of caps unless you handle it.  (Contrary to Power recommendations for this project.) In theory there are tricks you can do to manipulate high peak power into stead average power. However there are a few things you have to keep track of.

Realistically I would try to keep the load under 25 watts here. Those batteries even at 1C mean you had 132W which could be consumed by a whole panel. This is why the third panel started to flake. In the demo you were not lighting all the LEDs which reduced your peak current. Solid white panel would have done it on a single probably. (I do not recommend trying it!)

Overall cool project. I liked some of the animations and stand. Was curious how you charge it or if there is a thermal problem. Note I have ignored the Pi's power from consideration.

daveythacher, 2020-10-12:
If the batteries had an over discharge/over current circuit you could potentially wired the batteries straight do panel. The boost would be needed for Pi. This will mean you will power your panels by 4.2 to 3 volts. However the panels will likely stop working around 3.2-3volts, or start to dim noticeably. 

You will lose white color correction due to the IV plot and reduced main voltage. However the LEDs will still light and in software you can correct some of the this. This will save a decent amount of current from the batteries while reducing load.

Boost step up current draw from battery would be more aggressive as battery voltage goes lower. ((5/4.2)/0.92)*I to ((5/3)/0.82)*I

It is doable, just a little more annoying.

mikecann, 2020-10-15:
@daveythacher it definately sounds like you know what you are doing mate, much more than I. I have no idea how I managed to get it to work considering the little I know about this sort of thing. Should I ever attempt something like this again ill be sure to look you up! :)

daveythacher, 2020-10-15:
I look up your battery and found this:
The battery is rated for 2C continuous discharge, however the wiring is only rated for 1A so be careful.

This puts you at 2*4*6=48A from battery. So you have more than 46A needed. However you have issue with boost (problematic only when battery discharges below certain) and wire. The wire power may have be okay for average power, but I doubt it.

This mean your batteries unless you run them to complete discharge for max peak load, your batteries will be stable. However you could be heating up the wire. 2C is a decent discharge, which is probably why it works. I still think there was a few things that could have been done.

You would want to show more caution against fire hazards. LiPo's can become volatile if taking outside spec. Please note some of this is for your benefit, but also for others benefit. Documentation here is very poor. These represent a weak form of documentation.

mikecann, 2020-10-16:
@daveythacher thanks for some important tips there mate. I was aware of temperature issues inside the cube, TBH I dont usually run it for long enough for it to become a big problem. If I was at a hack day or running a long demo or something I would have to put some more serious thought into that. 

Same goes for the wires. You are right I probably should be concerned about taking them past what they are rated for. Im hoping that the short periods of use negate this somewhat.

Definately good info for anyone that reads this issue in the future tho!