hzeller/rpi-rgb-led-matrix #991: DietPi Buster + Raspberry Pi 4 works well!

raspberry-pi-maker, 2020-02-22:
When I used Rasbian Buster with Rpi4 issue #483 occcurs.  I have to use an old Rasbian Jessie. But the Rpi 4 doesn't support Jessie. As a result, I have to use Rpi 3 instead of 4.
Yesterday, I tested DietPi Buster with Rpi 4. The RGB LED matrices display is good. No flickering.
I've write the test process in my blog (https://iot-for-maker.blogspot.com/2020/02/led-5-raspberry-pi-4-dietpi-buster.html).  I hope this can help you who want to use Rpi 4.

mikecann, 2020-02-26:
This is good to know. Im about to embark on a project involving 6 x 64x64 panels using a pi 4 and the Electrodragon HAT. 

I was hoping to power them from battery, how likely do you think that is?

raspberry-pi-maker, 2020-02-26:
When all the RGB LED pixels are on, max power might be needed.
However, this rarely happens. 
Of course, it's a good idea to have a power supply that can handle the maximum power.
In my case, considering the cost, there was no big problem using 5V10A to handle P3 64X32 4EA.

marcmerlin, 2020-02-27:
Thanks @raspberry-pi-maker I was about to ask @hzeller if there was any benefit to using an rPi4 over a 3?
From what I understand the rPi3 is already able to output faster than the panels are able to absorb the input and that while 256x256 is possible with an electrodragon active-3 board, it's pushing the limits of what can be done with 3 channels/1 Pi (refresh rate and color depth are not great).
My assumption so far is that an rPi4 doesn't change this and 256x256 is still more or less the limit.
I opened 
https://github.com/hzeller/rpi-rgb-led-matrix/pull/971  https://github.com/hzeller/rpi-rgb-led-matrix/pull/971/commits/57031074130bfe49dc2ed9d0191febaf0ab364d1 to write this to the best of my understanding.

mikecann, 2020-02-27:
@marcmerlin I actually found that the pi4 was cheaper than the pi3 thats why I chose it. Plus the extra power lets me run more complex code or run in a higher level language or something.

hzeller, 2020-02-27:
Yes, the faster internal processing makes a Pi4 more desirable than a Pi3 even if the clock to the matrix is at their limit.

raspberry-pi-maker, 2020-02-28:
I agree with @hzeller . 
Even though the rPi3 has enough processing power to GPIO handling, making the contents to display is also important.
I like to use OpenCV to create the contents. The greater the processing power, the faster the OpenCV's processing speed. Therefore,  rPi4 is more advantageous than rPi3 when working with OpenCV such as video display.