hzeller/rpi-rgb-led-matrix #993: Red bar when using 3 daisy-chained

rijotech, 2020-02-26:
I managed to get 9 64x32 Led matrix attached to work using 3 daisy-chained connectors but at the last bottom I'm getting a red bar I can't seen to be able to get rid off. 
I tried grounding the E line but still nothing. 
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3
Any suggestions? 

[image]

shanerooni, 2020-03-02:
how are you powering these?  are they tied to a transformer?  i'm going to be setting up 12 as a 3x4 with 32x64s, and the power has me a little concerned.  i'm planning to get a 5V 60A transformer, and directly wiring 6 of the 12, and chaining the power as couples from those 6.

rijotech, 2020-03-02:
I'm using a 5V 60A transformer and only have 9 out off 12 32x64 panels connected, that should be enough power, I have tried grounding the E and many other things but can't find a way to get rid off the red bar. 
I'm going to try feeding power to the line with the red bar from another power adapter but I doubt that may be the problem.
[image]
[image]
[image]

shanerooni, 2020-03-02:
are you able to run this all from one pi 4?  i have the adafruit hat, which will only allow for one chain(?), i guess.  i'm still waiting on the panels to test them, so i only have 2 right now.  can i connect 12 panels to one pi 4 with the single panel output form the adafruit hat, or will i need the same setup you have?  appreciate the help!

rijotech, 2020-03-02:
I'm running everything from a raspberry pi3 using Hzellers board and 3 chains using all 3 daisy, Honestly speaking I'm not an expert on this I'm basically learning like some people in here but so far I can say i have been making some progress since I knew 0 and now I have some knowledge but lot of stuff is a bit confusing to me still, I have used the Adafruit Hats but I like the Hzeller Hat better for obvious reasons to me it works better also since in my case when running multiple panels on a single daisy distort the image I'm loading even after applying , I developed a mobile app to remote controller the display from my phone and create playlist, schedule what to display and a lot more but my only problem so far is finding a way to get rid off that red bar.

shanerooni, 2020-03-02:
i was planning to do the same thing with trying to create a lightweight custom app.  i'm not doing anything fancy or intensive like displaying a video or large image - more just a running stopwatch and scrolling text - so hopefully i can run it off of the hat and pi 4 since i already soldered that crap together haha.  i also posted the issue to see if i can have multiple canvases to update separately at the same time.  are you able to connect the pi to the phone without being on a shared wifi?  i also started with 0 knowledge a few weeks ago, so i'm getting there.  thanks again!

rijotech, 2020-03-02:
LOL I made the same mistake when I started learning this, I feel your pain, I managed to create a bluetooth PAN sharing connection with a static ip at the raspberry pi leaving the WiFi open, the connection is a bit  slow when loading the images but everything works like great the only problem I'm facing is that I have to catched the images once to load faster the next time I open the app.
  
I control everything via web app PHP, Javascript, jQuery and shell scripting.
 
[image]

I even created a method to remote control it using Siri voice command in any language and a wireless bluetooth panic button.

I want to create a panel for the groceries owners to see if they can use it to alert authorities in conjunction with a panic button and mobile app so that when they are getting rob the panel will display a Alert to notify people outside so they can alert the police.

hzeller, 2020-03-02:
I suspect the red stripe is from some stuck data line.

To isolate the problem try to use a different panel as the first panel in the lower section that now has the strip.
   * If it goes away: the first panel is broken and replacing it will likely also fix the red strip in the subsequent panels in that line
   * if it persists, then there is something with the wiring on the adapter board: closely inspect for soldering bridges or open connections there.
  
I have also once accidentally fried a GPIO pin on a Pi; computer kept working but one GPIO pin was not working anymore. So if the above is not resulting in anything, try also a second Pi if you have it at hand.

shanerooni, 2020-03-02:
holy crap! would you be willing to share the app?  it seems like your trying to turn these into a business venture, which is great.  i could probably figure out how to customize the app for what i'm doing.  it looks like you're using an iphone, though, and i'm on android - but being a web app it shouldn't be a problem.  my email is on my profile if you want to take it offline so we don't keep conversing on this string.

rijotech, 2020-03-02:
> I suspect the red stripe is from some stuck data line.
>
> To isolate the problem try to use a different panel as the first panel in the lower section that now has the strip.
>
> * If it goes away: the [...]

I tried all suggested possibilities and more, I replaced the panels, switched data cables,  replace raspberry pi, replace LED hat, inspected the boards to make sure there is no excessive soldering, recompiled the project with the latest codes, but still the same problem.

This is the command I'm implementing to display the image.
sudo led-image-viewer usa.gif  --led-rows=32 --led-cols=64  --led-slowdown-gpio=4 --led-chain=3 --led-parallel=3

Could it be I may be missing something in the command line?

hzeller, 2020-03-02:
I suspect you already went through the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting), but just in case this was missed:
the pins are also used by `SDA` and `SCL` on the GPIO header. Did you make sure to switch off i2c in your Pi configuration ? Is there any program trying to use the i2c in parallel ?

rijotech, 2020-03-02:
> holy crap! would you be willing to share the app? it seems like your trying to turn these into a business venture, which is great. i could probably figure out how to customize the app for what i'm [...]

Even If i share the mobile app It won't work for you the way it works for me since the app is just a remote control, most of the magic happens in the raspberry pi.

shanerooni, 2020-03-02:
no worries.  i do dig it, though.  i probably don't need something that complex, but i do need to try and figure out how to connect the pi to the phone as you described, and get some basic web app working.

hzeller, 2020-03-04:
Did you check your i2c settings ?

rijotech, 2020-03-04:
> i2c
That's the only missing thing I have not tried yet and yes I'm using the clock to schedule custom commands via crontab, I will check that today.

> Did you check your i2c settings ?

That's the only missing thing I have not tried yet and yes I'm using the clock to schedule custom commands via crontab, I will check that today.

hzeller, 2020-03-04:
it should be sufficient to use the RTC at boot time to sync the system clock, then switch off that interface.

rijotech, 2020-03-04:
Yep you was right it was the i2c switch on the LED board.

closed 2020-03-04 by rijotech

rabbitmaiden, 2021-01-23:
Just commenting for future people running into this: if you bought the board from Electrodragon and are using the third chain there's a switch on the board you need to flip to avoid this issue.