hzeller/rpi-rgb-led-matrix #995: Light specific LED in matrix (x,y)

JoeMevo, 2020-03-02:
Hey everyone, I'm trying to avoid doing an image hack where it just changes the image based on what I want. I want to fill the led board like a thermometer fundraising type ordeal. If 10% of fundraiser fill (x,y)-(x,y). Is there a way to specify what LEDs based on location to light up? I'm not finding anything specific in the docs. Haven't jumped too far into the example API's but if someone knows a specific example in one of those that would be great too.

Using a 16x64 (2x 16x32 boards from adafruit)

hzeller, 2020-03-02:
the canvas the LED-matrix has a SetPixel(x, y, color) method that allows you to set any pixel as needed

JosephC1337, 2020-03-02:
Very nice, thank you so much. I figured there was but wasn't able to find something on it and didn't want to misinterpret it. Great work you've done here :)

hzeller, 2020-03-04:
The API is the point where you could start reading
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h

Also the example codes in examples-api-use/ might be useful:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/minimal-example.cc

JosephC1337, 2020-03-04:
So thanks for the help. So right now I'm using Python and have it where I scrape a value from the web then calculate the total % of fundraiser complete. Then it fills the gauge based on the % returned. This is all functioning great. I'm trying to avoid just using images and change IMG based on quality, really want to use nothing but code where I can update the parameters for future use. Once I get this final part done I'm more than happy to share how I accomplished what I have thus far.

**So now for my question**
Say I want to use a variable for text and have it print in ONLY this area. 

        font = graphics.Font()
        font.LoadFont("../../../fonts/7x13.bdf")
        textColor = graphics.Color(255, 255, 0)
        pos = offscreen_canvas.width
        my_text = self.args.text

[image] (I Livestream my board so I can work on it remotely from home)
How do I tell it to only print that area? SetPixel was easy to use in loops to make a gauge based on the size of my canvas. However, I don't see how I can create a "secondary" canvas so to speak to just print the input text if I wanted to. Don't see any examples to take inspiration from. I see offscreen_canvas.width/height often but correct me if I am wrong that is for the entire canvas.

hzeller, 2020-03-15:
You can use `drawText()` in the same canvas (see the [graphics example](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/graphics.py).
That functions takes an x and y coordinate where you can tell it where to write things.
If you want to clear the area, you can draw a black rectangle over that (or maybe just write with black text the same text you wrote before).

If you want to do fancy things like clipping, you can implement a Canvas interface which delegates to some underlying canvas. This should be fairly straight-forward in C++, I don't know if the Python wrapper has the concept of Canvas being an interface, you have to explore that.

hzeller, 2020-03-15:
You can see an example of a delegating 'window' canvas in this comment:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/994#issuecomment-599158575

andrewsiemer, 2020-05-07:
> You can use `drawText()` in the same canvas (see the [graphics example](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/graphics.py) ).
> That functions takes an x [...]

Could you possibly help with an example to do this using the clipping method in python? I also cannot find how to draw a rectangle with the graphics example. Thanks in advance.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin