hzeller/rpi-rgb-led-matrix #996: rPi4 needs --led-slowdown-gpio=2, should it be a default?

marcmerlin, 2020-03-05:
~/rpi-rgb-led-matrix3/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0 --led-chain=1 --led-show-refresh --led-slowdown-gpio=2 --led-parallel=3   --led-pwm-dither-bits=1 --led-pwm-lsb-nanoseconds=50 --led-pwm-bits=7 --led-panel-type=FM6126A -D0
did not give me a workable display, this is using the code built on rPi3, which was a bit slower

marcmerlin, 2020-03-05:
[image]

marcmerlin, 2020-03-05:
With --led-slowdown-gpio=2, it works without issues.
As a bonus, ABCDE panels (128x64) in led-parallel=3, work a bit faster than on rPI3 (411.7Hz vs 373.3Hz)

hzeller, 2020-03-05:
yes, i am considering making the default dependent on the detected pi.

closed 2025-11-27 by marcmerlin