hzeller/rpi-rgb-led-matrix #997: Question about power and capacitor wiring

gingters, 2020-03-08:
Hi. Since my last question I now got myself two 16x32 P10 Outdoor panels, and only then I read the wiring and "A word about power" section 🙈.

So, to be sure everything goes right I now ordered a new power source (5V 10A, 50W) as my current supply is not strong enough, some capacitors (3300µF/25V) and the suggested active connector board from china (2 boards for 3 USD each, plus 4,32 shipping, wasn't that bad after all).

Just to be sure, as my electronics knowledge isn't that good (as in, I never learned a lot, and when I did that was over 2 decades ago ;) ): You (@hzeller) suggested two of these capacitors in parallel, directly at the power connector on the panels.

My panel has one power connector with four pins, 2 VCC and two GND. I double-checked that both VCC pins are connected together and both GND pins are too, so I think we only have two of each to evenly distribute the power between both pins.

I would now solder one capacitor between the inner VCC and GND and the other capacitor between the outer VCC and GND (to make sure they don't get shorted out by crossing their connectors).

So that would look like

```
  ------ CAP1 ------
  |+               |-
  |      CAP2      | 
  |     |+   |-    |
VCC   VCC   GND   GND
```

Is that correct?

hzeller, 2020-03-08:
Yes, this looks good.
If you only have one or two panels connected, maybe your power supply takes care enough of the power surges so you might not even need the capacitors. The longer the cables and the more panels are connected, the more likely you need capacitors.

gingters, 2020-03-08:
Thanks for the fast answer!

This is great. Cables won't be long from power supply to panel, and I really hope to only need one panel per display, two at max. And also I don't plan on having them show full white a lot ;)

They should only show a few letters and numbers in red, green or yellow, sometimes the time and if I really need two for the width, then I maybe can display a bit of a scrolling text (that would probably be not very convenient /readable on only one panel).

gingters, 2020-03-16:
Before I open up a new issue, I'd like to reopen here and ask another question that - I think - fits well here:

Is there a way to determine upfront if my panel needs 5V signal input or if the 3.3V from the Raspi GPIO is enough?

That is: Should I wait until the active driver boards arrive or is there a good chance that I can start coding and testing without the board right away?

Also:
I now ordered a tad more beefy power supply for my two panels (https://www.amazon.de/gp/product/B07DQKM9P7/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1). The cables that came with the panel are 60cm from power supply to first panel and then from that connector they extend another 40cm to the second panel. Is that short enough to avoid the capacitors for the time being? (I'm a bit reluctant, as my soldering skills aren't that good.)

hzeller, 2020-03-16:
You can try connecting it [manually](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) to your pi and see if it works. There is no harm to try, it will just not work properly. Keep your data lines short though (~15cm) to not load the GPIO pins with too much impedance (the GPIO pins are directly connected to the internals of the processor).