hzeller/rpi-rgb-led-matrix #1001: received signal SIG33, Real-time event 33.

marcmerlin, 2020-03-10:
I have code that links in rgbmatrix.
With smaller array sizes, it works fine (up to about 200x192).
When I try bigger arrays (250x192), I get this crash
```
Thread 3 "Table_Mark_Este" received signal SIG33, Real-time event 33.
[Switching to Thread 0xb5f11450 (LWP 10280)]
0x00051fcc in rgb_matrix::GPIO::SetBits (
    this=0x7a108 <rgb_matrix::CreateMatrixFromOptions(rgb_matrix::RGBMatrix::Options const&, rgb_matrix::RuntimeOptions const&)::io>, value=131072) at ../include/gpio.h:58
58	    for (int i = 0; i < slowdown_; ++i) {
```

I build with 
LDFLAGS += -L$(NATIVE_ROOT)/rpi-rgb-led-matrix/lib -lrgbmatrix -lrt
```
rgb_matrix::RGBMatrix::Options defaults;
defaults.hardware_mapping = "regular"; // or e.g. "adafruit-hat"
defaults.rows = 64;
defaults.cols = 128;
defaults.chain_length = 3;
defaults.parallel = 3;
defaults.pwm_lsb_nanoseconds = 50;
defaults.pwm_bits = 7;
defaults.led_rgb_sequence = "RBG";
defaults.panel_type = "FM6126A";

rgb_matrix::RuntimeOptions ropt;
ropt.gpio_slowdown = 2;

rgb_matrix::Canvas *canvas = rgb_matrix::CreateMatrixFromOptions(defaults, ropt);
while (canvas == NULL) Serial.println("Canvas did not initialize");
matrix->setCanvas(canvas);
```

Seeing slowdown, I commented out "ropt.gpio_slowdown = 2;" and now I get this:
```
Thread 3 "Table_Mark_Este" received signal SIG33, Real-time event 33.
[Switching to Thread 0xb5f11450 (LWP 11554)]
0x00052004 in rgb_matrix::GPIO::ClearBits (
    this=0x7a108 <rgb_matrix::CreateMatrixFromOptions(rgb_matrix::RGBMatrix::Options const&, rgb_matrix::RuntimeOptions const&)::io>, value=0) at ../include/gpio.h:64
64	  inline void ClearBits(uint32_t value) {
```
And if I lower the resolution back to something smaller, things work again:
```
    const uint16_t MATRIX_TILE_WIDTH = 200;  // not 250
    const uint16_t MATRIX_TILE_HEIGHT= 192;
```
I realize the error may be in my code (which I can't quote here, it's big), but I don't understand SIG33 is the rgbmatrix lib

hzeller, 2020-03-11:
never seen this; is this something you only see when running in a debugger?

marcmerlin, 2020-03-11:
No, without the debugger, I get SEGV.  
I missed the part that non crashing code gives out SIG33 reliably every time I run under gdb on rPi4.
Maybe it's a Pi4 issue. Let me downgrade to Pi3 and try.

hzeller, 2020-03-11:
do you get the same problem if you run the basic demos with the same resolution?

marcmerlin, 2020-03-11:
Yep, it works under rPi3. So I guess gdb is broken with on rPi4, at least with whatever the library does to get realtime. Sorry for the false alert.

marcmerlin, 2020-03-11:
Then again, I still get a crash in rgb_matrix on rPi3. I built my code with electric-fence to hopefully force a SEGV in the correct section of code, but it's still happening inside rgb_matrix.  
I realize it's not fair for me to ask you to debug my code even if gdb happens to find a SEGV in your library, but would you have a suggestion of how I can better track down the issue?
```
Thread 3 "Table_Mark_Este" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7601d450 (LWP 3678)]
0x0004e24c in rgb_matrix::GPIO::SetBits (this=<optimized out>, this=<optimized out>, value=131072)
    at ../include/gpio.h:66
66          *gpio_clr_bits_ = value;
(gdb) bt
#0  0x0004e24c in rgb_matrix::GPIO::SetBits (this=<optimized out>, this=<optimized out>, value=131072)
    at ../include/gpio.h:66
#1  rgb_matrix::internal::Framebuffer::DumpToMatrix (this=0x769eefd8, 
    io=0x723dc <rgb_matrix::CreateMatrixFromOptions(rgb_matrix::RGBMatrix::Options const&, rgb_matrix::RuntimeOptions const&)::io>, pwm_low_bit=<optimized out>) at framebuffer.cc:690
#2  0x0004a49c in rgb_matrix::RGBMatrix::UpdateThread::Run (this=0x76a7df18) at led-matrix.cc:99
#3  0x0004ec00 in rgb_matrix::Thread::PthreadCallRun (tobject=<optimized out>) at thread.cc:25
#4  0x76c5bfc4 in start_thread (arg=0x7601d450) at pthread_create.c:458
#5  0x76be8038 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:76 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

marcmerlin, 2020-03-11:
Ok, it was a buffer overrun in some code I included and somehow eletric fence didn't catch it when it's exactly what it should have done.
After it scribled elsewhere, it caused unrelated code (yours) to crash

closed 2020-03-11 by marcmerlin

marcmerlin, 2020-03-12:
And to share the knowledge I just learned:
- electric fence does not working against global arrays
- ASAN does, and as luck would have it, I have coworkers who worked on it :)

It's slightly broken on arm64 right now, so you have to use it like so.
1) add this to makefile
CFLAGS += -fsanitize=address
CXXFLAGS += -fsanitize=address
LDFLAGS += -fsanitize=address

2) run binary with 
LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libasan.so.3  ./Table_Mark_Estes
this is a temporary hack to get around an issue with asan on arm, but it works and will find exactly where your overrun, is.