hzeller/rpi-rgb-led-matrix #998: 64x64 Matrix with ICN2037 and High Standby Current

bleckers, 2020-03-10:
I have a handful of 64x64 panels with ICN2037 LED drivers (along with ICN2012's) and each panel draws about 100mA @5V when not clocking in any data to them (even when explicitly pulling the OE high and latching in the OE; "blanking operates on the latch").

I am able to "just" run these panels at 4V (at 3.9V they start to flicker) and they maintain the same 100mA draw (while reducing the overall power).

Is there any way to reduce the standby/quiescent current of these displays at all? Is there is a sequence I'm missing? Or might I just have to control the power supply to these in order to turn them off completely?

hzeller, 2020-03-15:
I am not aware of anything. Is it doing the same when you clock data into it, but just 'black' ?
If not, then maybe they have some poor off-current circuit somewhere. Maybe check with a infrared camera where the 500mW turn out to end up ?

Of course, there could be some shutdown sequence - for that I suggest to find the datasheet and see if they provide something like that.

daveythacher, 2020-11-12:
I have been puzzled by this too. Normal CMOS circuits will draw maybe a single mA at low clock or idle. (Usually uA(s)/nA(s) depending on size and type.) Clock speed is really nasty to power consumption, really bad if it requires a voltage step. Many panels uses a few basic ICs. IO buffers, constant current LED drivers and row select logic.

Most of the CMOS, row select and IO pins loads are capacitors. So basically if the pins change there will be a little power required. However once idle there really should be only a small draw per IC.

The answer for where the bulk of the power could be is in constant current drivers. These likely have some kind of feedback loop or reference current that is potentially increasing the idle power consumption. The may use a slightly less efficient circuit to save cost on fab process or die space. Most of these panels are not expected to be used in power critical environments. The LEDs themselves will use way more power. You would need many panels for that power consumption to seriously matter.

This is speculation so take it with a grain of salt. However I think one can built a current limiter using a reference current that then a current follower replicates against known current gain. This reference current could be a few mA per IC, which means there could be around (columns / 16) * 6 or 12 for 32x16 RGB matrix. 24 for 64x64 assuming 1:32 scan using two RGB chains.

100mA / 24 gives you around 4mA. So you could be looking at a reference current of around 3mA.

In the ICN2038S there is Iout = (1.24 / R) * 15. So for 20mA Iout you would have around 1.33mA moving through the Rext pin. The internal CMOS and comparator logic power consumption would increase the total draw for each of those 24 ICs. Now granted each color will have different Iout values to white balance at 5 volts.

There is also leakage, pull-up, etc. power loss.

If the are secretly FPGAs that could cause higher power draw compared to ASIC. If secretly using high internal clocks that could cause power draw. 

This one of the reasons software has really struggled with power consumption. They keep it in the high clock domain. Fab process is normally how they grow circuits and lower power consumption. Fab process has high cost so they try to generalize hardware which creates software.

Now once the fab process got a certain point they could start changing things. They just choose poorly. The programming model did not evolve. Therefore they are now screwed. Mobile market was one sign of trouble. They misunderstood the market and the options at their disposal. They were not prepared for the internet...we all know why now. (They be on the special path of limited means.)

daveythacher, 2020-11-12:
Now if you really want to have some fun. Clock a memory mapped LED driver with a fixed and non-changing number of LEDs on. First use low speed clocks, then high speed clocks. Mind you this is just to drive a the same image. I think I saw the power consumption on the panels jump 100mA for two panels. Granted I had the current gain dialed back.

So if I had to guess I would say my power consumption was a function of: Idle power plus LED power plus switching power. I can change mostly everything but idle power. Now this does not surprise me in college I took a course in VSLI. From what I remember, I think most of this checks out.

daveythacher, 2020-11-12:
Note I would not expect this load to significantly scale with voltage. More than likely this is linear or constant. One example of linear power consumption against voltage is linear voltage regulator. 

Now you can make a linear current regulator out of a Zener diode. These are usually very poor voltage regulators due to high power loss. However they may decent and cost effective voltage references. So if you want to ensure that the current limit resistor always has 1.24 you would get a 1.24 Zener diode in parallel with it. The Zener will take the additional voltage via some current to ensure this. This creates a voltage divider (sort of) but generates a little bit of heat.

If they are using that you would have some current flow through the Zener and resistor. In addition to CMOS, comparator, base current of transistor (assuming BJT), etc.

Linear is less efficient but is a lot cheaper and a Zener diode is easier to make in silicon. There is a lot of games they can play to control current, heat, cost, etc.

Again take all of this with a grain of salt.

bleckers, 2020-11-13:
I kind of worked around it by just switching the panels off with a MOSFET driver (and making sure to disable the output level converters as well, because these drivers don't seem to isolate from there either and will turn on the displays through the level converter's logic voltage). This removed standby current when the display isn't used at all, but it doesn't stop the fact those driver chips are eating half a Watt per panel when operating. I think you're right @daveythacher, they are just using standard off the shelf processes and causing a huge amount of redundant wasted power over the life of all the panels created. It's just like poorly designed power supplies in mandated smoke alarms, over the span of millions of this, you need a power plant just to handle the power supply inefficiency.

My original requirement was to get standby power low, so that's solved for me, but it still doesn't solve the fact there's a tonne of wasted power with these drivers (happens on FM6124/RUC7258 based panels as well, so I guess it's systemic).

daveythacher, 2020-11-13:
> I kind of worked around it by just switching the panels off with a MOSFET driver (and making sure to disable the output level converters as well, because these drivers don't seem to isolate from [...]

I forget how I explained away the leakage power from inputs when the panel is not powered. I know there is red LEDs which will light dimly. I know of several ways this can happen: leakage, diode/transistor/mosfet body/well leakage, cap leakage (maybe), protection clamp diodes, BJT transistors, transistor configurations/layout/architecture issues, etc.

It would depend on many things but likely it has to do with how it is put together. Some of the control logic probably floats in this case in a manner that allows very small current flows. Which is enough to light some of the LEDs. Normally if there was power applied in other sections this would not happen.

All of this could have been prevented but this would increase cost slightly. These panels are highly cost optimized. Just not operation cost optimized. Like you said if you consider the loss of money from having to pay for wasted power across a large time interval. That is where efficiency really shines. over the long term.

Its funny we talk about this but ARM likely had something like this happen. It was magical and amazing. Super low power CPU. I shake my head at ARM all the time for pretty much everything under the sun at this point. I have appeared to talk up ARM on a few occasions but never really used on in my personal projects...hmm.

Lot of people do understand VSLI is kind of a world of its own. It still plays by all the same rules, but they can be unforgiving and do some crazy stuff. Like run tons of transistors on next to nothing. It can also do the opposite and stuff in the middle. Silicon economics has kind of been a potential problem for a while. Most people just don't get that one either.

daveythacher, 2020-11-13:
These panels use 74HC245 and 74HCT245 for buffers to try to fight issues with signal loss. The 74HCT245 is TTL compatible. The standard application is that these will be used at 5V. They have hit or miss support for 3.3V. The 74HCT245 is supposed to work only at 5V. 

The ribbon cable seems to me to be cheating transmission line behavior. The FPGA receiver cards use termination on the outputs. However the panels do not. Maybe it is not needed? It appears to work in the setups I tried.

This projects uses them as logic converters. It is also not really a level shifter, however the 74HCT245 does appear to possibly have a slight overlap which may allow it. These are mostly here to buffer the Pi outputs. This project includes no termination.

If you clean up the power line with decoupling it my be enough for small chains. If not blasting the clock rate. However some of the faster Pi models may be moving into faster speed domains now. Factor in larger setups tend to use more panels which can cause power supply issues. You have a recipe for disaster.

daveythacher, 2020-11-13:
I guess the PCB trace could be the termination in the panels.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828