hzeller/rpi-rgb-led-matrix #999: horizontal offset issue with rpi-rgb-led-matrix/utils/video-viewer

marcmerlin, 2020-03-10:
When playing videos, I found that some of them (not all) are offset with a black far at 90% on the right, and then the left part of the video on the right side of the screen
Source video:
VIDEO:  [WMV2]  320x240  24bpp  1000.000 fps  632.0 kbps (77.1 kbyte/s)
Playing with:
~/rpi-rgb-led-matrix/utils/video-viewer --led-gpio-mapping=regular --led-rows=64 --led-cols=128 --led-row-addr-type=0 --led-chain=3 --led-show-refresh --led-slowdown-gpio=2 --led-parallel=3 --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A --led-rgb-sequence "RBG" $i
(output is 384x192)

marcmerlin, 2020-03-10:
[image]

marcmerlin, 2020-03-10:
Another video with the same aspect ratio (640x480) however plays perfectly.
VIDEO:  [WMV1]  640x480  24bpp  1000.000 fps    0.0 kbps ( 0.0 kbyte/s)

marcmerlin, 2020-03-10:
when it works, though, the result is cool (shot with a camera where I could lower the framerate so as not to get the black bars that would be very visible at 100Hz)

marcmerlin, 2020-03-11:
when it works, though, the result is cool (shot with a camera where I could lower the framerate so as not to get the black bars that would be very visible at 100Hz)
https://www.youtube.com/watch?v=wD5KXFJFuO0
(added missing video of maybe the highest resolution rpi-rgb-led-matrix display so far)

hzeller, 2020-03-15:
I have not seen this problem; do you have an example video that can be shared that shows the problem ?

I can only suspect some sort of mess-up with the internal way the AV frame-buffer represents data vs. how we copy it ? The pixels are set here; it 'looks' like this should be correct, but maybe I am missing something ?
https://github.com/hzeller/rpi-rgb-led-matrix/blob/cf3c3c2ef0b1d1a637c1b9590965186a05142f5c/utils/video-viewer.cc#L51-L61

I'll play with some videos of different aspect ratio.

hzeller, 2020-03-15:
I've fixed that the aspect ratio is properly maintained, maybe this was the issue with the video in question ? Can you try with the current code ?

marcmerlin, 2020-03-15:
I'm sure it was a ratio issue indeed. I'll send you a test video (I took down my 384x192 display to try and get 384x256 working, although I may give up on that
I'll send you a link via Email

hzeller, 2020-03-15:
I think I fixed it, please test with latest code.
In order to force the original aspect ratio that triggered the problem, use the `-F` flag (for 'full-screen').

closed 2020-03-15 by hzeller

marcmerlin, 2020-03-16:
will test as soon as I get my panel array back in working order, thank you.

marcmerlin, 2020-03-18:
I just finished getting my 384x256 panel working and can confirm your fix works great, thank you.

hzeller, 2020-03-18:
Wow, 384x256 is probably the largest I have heard anyone put together!

At this point, filling the full panel with pixels might already be a limiting factor (due to the way the bits have to be sliced and diced, SetPixel() bottoms out at a few megapixels/s - so with 100k pixels, that might already cut into 'do animations with 60Hz' desire).

If you use a pre-computed stream ([content-streamer.h](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/content-streamer.h)), this can be of course much faster as the stream is stored internally in pre-sliced form (but uses a lot more memory than just a simple RGB framebuffer).

marcmerlin, 2020-03-18:
@hzeller I think 384x256 will probably stand as the upper bound because honestly beyond that, the flickering will be painful. I already had to use my (real) camera in manual slow shutter mode to get proper pictures/video from the display as it was impossible with a cell phone.
Believe it or not, but while the rPi3 was indeed struggling to decode videos and render the, the rPi4 managed.
Yes, I know about pre-computing, but demo videos, I was curious to see how it would work with decoding. 
Output if you are curious: https://youtu.be/cP3h6boA3UQ
The video looks slow because I lowered the shutter rate on my camera to avoid showing the refresh bars given that the refresh rate is only 100Hz or so.

commit eb3027d0 (2020-03-15): Allocate correct size for larger-than-input matrix. Fixes #999