hzeller/rpi-rgb-led-matrix #1002: Rust bindings are out-of-date

dabch, 2020-03-11:
Hi everyone,
the rust binding by @Helvethor  linked in the readme seems wildly out of date by now. It was last updated two years ago. Upon trying to instantiate a Matrix, it simply segfaults. 
I am currently working on building Rust bindings for the current version of this library. I'm a Rust newbie, but taking some inspirations from the old bindings and lots of trial and error will get me there, one day. If so wished, I'd be happy to supply my bindings for you to link. Please tell me, and I'll open a pull request with a link as soon as they are ready.

hzeller, 2020-03-12:
yes, please do send a pull request.

jannikkeye, 2020-04-06:
@dabch Hey, have you made any progress? :)

dabch, 2020-04-06:
@jannikkeye yes I did! See dabch/rust-rpi-rgb-led-bindings
They're not fully finished yet, I wanted to also add some docs and test them some more. Still plan on doing that in the next few weeks but I got caught up with other stuff. But please do try them out

jannikkeye, 2020-04-06:
@dabch Awesome, will do :) I've been patiently waiting.

TDHolmes, 2020-08-30:
Hey all, I (and @sphinxc0re) have started maintaining the original crate and have updated it. I'm actively working on it in my spare time, and would love contributions as well! [Here is the new repo link](https://github.com/rust-rpi-led-matrix/rust-rpi-rgb-led-matrix).