hzeller/rpi-rgb-led-matrix #1004: Flickering when fired by local tty, less flickering when fired over remote ssh...

leovanderveen, 2020-03-14:
I've got a simple bash script that fires a python script that drives my 96x64 led matrix at boot time. I have this strange issue that the screen flickers when started by the Pi itself but the flickering is less when the bash script is ran by a remote computer via ssh. It's not just a startup issue cause when I kill the remotely triggered process and restart it locally from the Pi itself the flickering again is more visible. Any ideas?

- raspberry pi lite distribution
- adafruit-matrix-hat with hardware modification
- followed all the suggested troubleshooting tips.

hzeller, 2020-03-15:
Maybe your local shell will impose some more memory churn inside the Pi when it displays things on a screen. In particular if you have a windowing system running this can very well be already too much for the Pi. In general I suggest to always use the Raspberry Pi headless.

I suppose you already did the `isolcpus=3` setting ?

From what I hear from people is that https://dietpi.com/ usually produces less visual flicker.

leovanderveen, 2020-03-15:
Alright, yes i did the isolcpus=3 setting. I will try a headless setup, is there any setting I have to make to not force the HDMI output even when no plug is inserted?

I'll also try the dietpi distribution

leovanderveen, 2020-03-15:
It's a model 3B+ Raspberry Pi, so that shouldn't be the problem i guess

hzeller, 2020-03-15:
I have rarely connected a monitor to any of my Pi's and they are perfectly happy without a HDMI connection.