hzeller/rpi-rgb-led-matrix #1005: Random artifacts in video appear periodically

jabulonboaz, 2020-03-15:
When i play back any video or any vid.stream i get artifacts appearing in every video that appear for a millisecond then disappear and appear again randomly within a video say around every 4 seconds. Pls see pics - two showingthe issu, one normal. What causes these and how can i get read of them pls?
I am using a p3b+ with an Adafruit RGB matrix bonnet, four 64x64 P2.5 panels in series and a 5A 10A PSU. 
[image]
[image]
[image]

hzeller, 2020-03-15:
if it is regularly, it sounds like the symptom of one-wire connection not disabled
https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting

If that is not the reason, try slowing down gpio with the `--led-slowdown-gpio=2` (or even higher) option.

Finally, sometimes glitches can happen if the power cabeling is too noisy. Make sure to have power wires individually connected to the power supply, have them thick, potentially add a capacitor etc.

But from the described symptom it sounds like one-wire connection to me.

hzeller, 2020-03-15:
Also, have you a monitor connected to the Pi ? Running a graphical UI also is known to mess with the timing of the outputs. I recommend to always run the Raspberry Pi headless.

For a slim raspberry Pi operating system I recommend DietPi ( https://dietpi.com/ )

jabulonboaz, 2020-03-15:
> Also, have you a monitor connected to the Pi ? Running a graphical UI also is known to mess with the timing of the outputs. I recommend to always run the Raspberry Pi headless.
> 
> For a slim raspberry Pi operating system I recommend DietPi ( https://dietpi.com/ )

Hello thanks for the fast response and this amazing work you have done. 
I am afraid non of the above, 1 wire is disabled (i just checked again both in config.txt and raspi-config GUI) and also no monitor conected, i have installed the lite distribution you recommended on the instructions. 
All my wiring is fine is like you recommend but i have not used a capacitor, how many farad and do i put it in parallel with VCC and GNG right? 
Pls also see the additional picture that shows my CLI maybe i missed something there ? 
Any suggestion is welcomed.

jabulonboaz, 2020-03-15:
Fixed thank u! It was the gpio slowdown! I put it to =2 and the issue is gone. Thanks HZeller.

hzeller, 2020-03-15:
Raspbian Lite is good, but from the feedbck I get from users dietpi seems to be better w.r.t. not randomly generating flicker reasons.

Next thing on the commandline I'd try is `--led-slowdown-gpio=2` (or 3) and see if that makes a difference. The fact that 'further away' from the connection looks garbled indicates that the matrix might be too slow for the Pi sending data.

(As for power supply: I'd probably put in the order of 3300μF on each panel if that turns out to be an issue)

jabulonboaz, 2020-03-15:
> Raspbian Lite is good, but from the feedbck I get from users dietpi seems to be better w.r.t. not randomly generating flicker reasons.
>
> Next thing on the commandline I'd try is [...]

Thanks for everything it was the gpio slowdown i put exactly what u said and its working perfectly now.

closed 2020-03-15 by jabulonboaz