hzeller/rpi-rgb-led-matrix #1006: Question: Is there a way to stream a YouTube URL?

jabulonboaz, 2020-03-16:
Hello and thanks again for this amazing git!
Is there a way to stream a YouTube URL pls and if yes can you provide the commands or some guidance pls?
I am using a p3B+ and an adafruit bonet.
Thanks again

hzeller, 2020-03-16:
If I want to stream YouTube directly I usually just use the [FlaschenTaschen server](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display) and then use `vlc` to read the video and stream to that display.

https://www.youtube.com/watch?v=kTl-aV11rME

You can of course also download the video and play with the [video-viewer](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer).

jabulonboaz, 2020-03-16:
> If I want to stream YouTube directly I usually just use the [FlaschenTaschen server](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display) and then use `vlc` to [...]

Thanks again for the fast response. I cloned F-T but i am not sure how to stream the YouTube URL locally from the Pi via VLC on the Pi? 
I understand from the instructions how streaming say a stored mp4 clip from a PC that runs VLC works as source and how to stream via UDP to the Pi. But how to do it all on the Pi without streaming from a PC to the Pi i am not sure.
(I also never used VLC to stream YouTube but i will Google it.)

hzeller, 2020-03-16:
I am not aware of ffmpeg (the video viewer program is using that) directly reading YouTube URLs. But there are command line utilities that help download videos from sites and then you can run them locally with the video-viewer.

Try to download a video size that is closest to your panel-size to minimize processing to be done on the Pi.

jabulonboaz, 2020-03-16:
> I am not aware of ffmpeg (the video viewer program is using that) directly reading YouTube URLs. But there are command line utilities that help download videos from sites and then you can run them [...]

Do you know what i am doing wrong pls? I get this error and i can't start the FT server. Could u pls kindly provide some example commands?
[image]

hzeller, 2020-03-16:
Did you compile it with `make FT_BACKEND=rgb-matrix` ?

hzeller, 2020-03-16:
You also might need to update the rgb-matrix submodule to latest if you use `--led-panel-type`
(or do a fresh recursive checkout, I just updated the flaschen-taschen/ to include the latest)

jabulonboaz, 2020-03-16:
> Did you compile it with `make FT_BACKEND=rgb-matrix` ?

I have now and the server is listening to port 1337 but i have another problem when i stream i get only a line. What am i doing wrong? I am trying to stream YouTube or one of the files that already play back fine locally on the pi.
[image]
[image]

jabulonboaz, 2020-03-16:
> You also might need to update the rgb-matrix submodule to latest if you use `--led-panel-type`
> (or do a fresh recursive checkout, I just updated the flaschen-taschen/ to include the latest)

How do i update what is the comand pls? I have cloned recursive originally 2 days ago.

hzeller, 2020-03-16:
you go to the subdirectory server/rgb-matrix, then checkout master, git pull.

So, if you are in the toplevel `flaschen-taschen` directory, say
```
cd server/rgb-matrix
git checkout master
git pull
```

jabulonboaz, 2020-03-16:
> you go to the subdirectory server/rgb-matrix, then checkout master, git pull.
> 
> So, if you are in the toplevel `flaschen-taschen` directory, say
> 
> ```
> cd server/rgb-matrix
> git checkout master
> git pull
> ```

Its all up to date

jabulonboaz, 2020-03-16:
> I have now and the server is listening to port 1337 but i have another problem when i stream i get only a line. What am i doing wrong? I am trying to stream YouTube or one of the files that already [...]

Could u help me with this pls?

hzeller, 2020-03-16:
Before you test with VLC, I'd try first the provided clients in the

```
cd client
make send-image
./send-image -h <ip-address-of-your-pi> -g128x64 ../img/invader.png
```

The `-g` will determine the geometry, pass here the size of you LED panel, because the default is set to the original flaschen taschen art display.

hzeller, 2020-03-16:
With vlc, you probably want to invoke the one without UI, so simply
```
cvlc --vout flaschen --flaschen-display=192.168.1.155 --flaschen-width=160 --flaschen-height=96 <videofile>
```

with the parameters for the IP address, sizes and videofile of course. The videofile can also be a youtube link.

As with the manual download (`youtube-dl`), VLC also has some script built-in that does the download; it might need updating. But of course, this is all depends on YT not changing the underlying structure as the videos are of course not meant to be downloaded.

jabulonboaz, 2020-03-16:
> Before you test with VLC, I'd try first the provided clients in the
>
> ```
> cd client
> make send-image
> ./send-image -h <ip-address-of-your-pi> -g128x64 ../img/invader.png
> ```
>
> The `-g` will determine [...]

Apologies i don't follow i have a windows PC and using putty to SSH into the Pi how do i get the pictures that are stored on my laptop to display in my led matrix?
[image]

hzeller, 2020-03-16:
Not clear what the problem is: the image magic library could not read the image, maybe you need to install more libraries, such as libpng ?

jabulonboaz, 2020-03-16:
> Not clear what the problem is: the image magic library could not read the image, maybe you need to install more libraries, such as libpng ?

Apologies can pls let me know what commands to type one by one?

jabulonboaz, 2020-03-16:
> With vlc, you probably want to invoke the one without UI, so simply
>
> ```
> cvlc --vout flaschen --flaschen-display=192.168.1.155 --flaschen-width=160 --flaschen-height=96 <videofile>
> ```
>
> with the [...]

This doesn't work either i get tons of errors if you can tell me what commands to type one by one it would be great 
[image]

jabulonboaz, 2020-03-16:
> Not clear what the problem is: the image magic library could not read the image, maybe you need to install more libraries, such as libpng ?

I found some instructions online to install this library i did and tried again but nothing gets displayed on my panels

hzeller, 2020-03-16:
vlc: look at the error message. It says that it can't output audio, which we expect as audio is disabled on the Pi. Try add the `--no-audio` option.

hzeller, 2020-03-16:
`send-image`: do you get the same error message as before ?

jabulonboaz, 2020-03-16:
> `send-image`: do you get the same error message as before ?

I never get any error with that just no image is displayed

jabulonboaz, 2020-03-16:
> vlc: look at the error message. It says that it can't output audio, which we expect as audio is disabled on the Pi. Try add the `--no-audio` option.

Ok will try

jabulonboaz, 2020-03-16:
> vlc: look at the error message. It says that it can't output audio, which we expect as audio is disabled on the Pi. Try add the `--no-audio` option.

[image]

hzeller, 2020-03-17:
Looks like your raspberry Pi installation is messed up, as it seems to assume pulse-audio. Also, you're invoking it with a YT-URL, I'd first use a standard video file before trying the thing that is more likely to fail (because YT content download restrictions).

I'd just run the vlc from your regular Linux machine and send the stuff over to the Pi. Or, if you have Windows, install vlc there. The Raspberry Pi is likely not powerful enough to decode the video anyway.

hzeller, 2020-03-17:
(I've tried running cvlc from my dietpi installation, and it is able to play a mp4 video via flaschen-taschen. But it has a lot of CPU problems when downscaling a large video.)

jabulonboaz, 2020-03-17:
> Looks like your raspberry Pi installation is messed up, as it seems to assume pulse-audio. Also, you're invoking it with a YT-URL, I'd first use a standard video file before trying the thing that is [...]

Ok so back to the original ask how can stream an mp4 video file from my VLC in my windows machine to the pi display? I get that line i showed u earlier and not video stream. Can u send me a ln example of the comands along with the settings u use on VLC on your PC pls? Thanks again for all of your support i truly appreciate it alot.

jabulonboaz, 2020-03-17:
> Could u help me with this pls?

If u can help me with this original question i would be more than grateful

hzeller, 2020-03-17:
You should first start invoking your `ft-display` with the same `--led-something` options you usually invoke when e.g. running the demo programs (you should verify by running in the rpi-rg-led-matrix the `demo` in `examples-api-use/`). If that demo works, the `ft-display` also should work. The line in the corner looks strange as if something else is not properly connected. Your `ft-display` commandline you had above looked good at first glance, so you have to narrow down the problem there.

Once that ft-server is running, you can try some very simple thing to send to it. For instance, go to the `flaschen-taschen/examples-api-use/` and run the simple-animation which doesn't need any other dependencies. You can compile it on Linux, but if your Windows system is set-up to compile it should work there as well.
```
cd flaschen-taschen/examples-api-use
make
./simple-animation 192.168.1.155  # or whatever the IP is where the ft-server is running
```

If that is not outputting anything (it should be little space-invader sprite) you first have to find the cause of that.

Once that is working, you should be able to run the commandline version of vlc (I find the UI too confusing and not necessary) with the options described earlier. I suppose the `cvlc` will run similarly under Windows as it runs on Linux.

jabulonboaz, 2020-03-17:
cd flaschen-taschen/examples-api-use make ./simple-animation 192.168.1.155 # or whatever...

Is this supposed to work inside the pi without streaming something from the PC? What i don't understand is, are these the only comands needed or do i have to run the commands for the ft server before on the pi? And the ip address should be the pi ip address right ? What do u mean the ip address where the server runs? U ether stream from a PC to the pi that is conected to the matrix or you run examples inside the pi.

hzeller, 2020-03-17:
*> Is this supposed to work inside the pi without streaming something from the PC?*

Yes, this can be run from everywhere on the network, including the pi itself.

*> What i don't understand is, are these the only comands needed or do i have to run the commands for the ft server before on the pi?*

These are commands to build the simple animation and then run it is to test if your pi is working as intended. You can run them anywhere on the network, but it might be the easiest directly on the Pi when this is the only Linux machine you have around.

*> And the ip address should be the pi ip address right ?*

Correct.

*>  What do u mean the ip address where the server runs?*

The server program you run is `ft-server` that run on the Raspberry Pi - it receives the image information and displays it on the matrix. So "where the server runs" is the Raspberry Pi.

*> U ether stream from a PC to the pi that is conected to the matrix or you run examples inside the pi.*

Anywhere on the network. You always have the `ft-server` running.
The `ft-server` essentially makes your LED matrix a networked screen. You can send stuff to it from your other machines on the network such as your Windows machine (where you can run VLC to send video), or locally on the pi from some other shell.

You also can use the layering feature of flaschen-taschen and have one video play in the background and have some text or sprite draw on top of the running video and you can send them from different network locations - they are composited in the `ft-server` to one picture sent to the matrix.

This one for instance was done with multiple layers compositing in parallel from multiple sources on the network: https://www.youtube.com/watch?v=XeJBcNevuJE

Here is some plasma demo sent via flaschen-taschen to the display:
https://www.youtube.com/watch?v=h3xY_QZ7CS4

jabulonboaz, 2020-03-18:
> You should first start invoking your `ft-display` with the same `--led-something` options you usually invoke when e.g. running the demo programs (you should verify by running in the [...]

Hello thanks so much for all of this help really! 
I tried the simple animation example but i get nothing on my panel. I tested my panel before i did this with led-image-viewer in .../utils and my stored videos play just fine as mp4 or .stream. 
What i am i missing?
[image]

jabulonboaz, 2020-03-18:
I run the ft server at the same time still i get nothing

hzeller, 2020-03-18:
If you compile the ft-server `make FT_BACKEND=terminal`, do you see at least a terminal output ?

jabulonboaz, 2020-03-18:
> If you compile the ft-server `make FT_BACKEND=terminal`, do you see at least a terminal output ?

When i run that comand on the left terminal it just changes line nothing else happens and now when i run the ft server i get just a black terminal where as before i got "listening on port 1337"
[image]

hzeller, 2020-03-18:
you should see the animation. If not, can you double-check that the IP address is the IP address of the Pi ?

jabulonboaz, 2020-03-18:
There something wrong i am doing with the server command or set up right? 
[image]

jabulonboaz, 2020-03-18:
> you should see the animation. If not, can you double-check that the IP address is the IP address of the Pi ?

Yes the ip is correct

hzeller, 2020-03-18:
If you run the `ft-server` compiled with terminal, it should show a black screen (with a couple of dashes at the bottom). That is the canvas area it will display in with a reasonable terminal.

Make sure to not use too large of a `-D` range as it will try to [show the canvas in your terminal]( https://github.com/hzeller/flaschen-taschen/tree/master/server#terminal) which will look funny if you specify a size larger than your terminal can show. Keep it at default for now

The animation will send to the IP address at the 1337 port. If you are running the animation on the same machine, you can also use `127.0.0.1` of course, as this is localhost.

If you send stuff from some other machine on your network, make sure to not have some accidental firewall rule somewhere that blocks UDP port 1337.

jabulonboaz, 2020-03-18:
I reduced D to 30x30 and i saw the line then run the simple animation and got 3.3fps on the boot of the ft terminal but still i get no image on my panel
[image]

hzeller, 2020-03-18:
ok good, so you're receiving the data (the fps indicates that the server receives the packets). Maybe wait a little as the animation wanders around and assumes a size of 45x35, maybe it is just out of frame. Or the terminal can't really deal with the characters sent (it requires a terminal that can display UTF-8 characters and modern color codes but not sure how well the terminal works on Windows; terminal programs are probably not Microsofts highest priority :) ).

If you see at least the fps, you now can also try to send something via VLC to the terminal (giving it the `--flaschen-width=30 --flaschen-height=30` size) and at least you should see the fps change.

Anyway, with this now confirmed that you receive stuff, you need to figure out why the led-matrix is not working. just one streak on a corner indicates that some clocking is not active, or cables are loose or something else is using the GPIO-pins. Double check with a rpi-rgb-led-matrix demo that indeed the current cabeling and set-up is ok and working.

To be sure, I suggest to remove or rename the flaschen-taschen directory, do a fresh recursive checkout and then do the compilation with `make FT_BACKEND=rgb-matrix`, and then run `ft-server` with the the `--led-...` command line options you use for a successful run of the demo have to give to the led-matrix programs (chains, parallel, etc.). (and of course make sure that there is no other program in the background running doing some matrix update or accessing the GPIOs)

jabulonboaz, 2020-03-18:
I tried sent-image and i get some colours on the terminal when i run one of your jpg and back terminal with 0.2fps when i run the invader.png  but nothing on the panel on any occasion
[image]
[image]

jabulonboaz, 2020-03-18:
I tried vlc with an mp4 video and this is what i get on the terminal but nothing on the panel.
I have checked the panel is fine it works perfectly with the demos and also i do not use windows terminal i am ssh in the pi what u see are the pi terminals. What maybe the codec i select on VLC is wrong? 
[image]

hzeller, 2020-03-18:
Yeah, looks like your terminal can't deal with the color codes, but the color salad shows that something was received :)

The broken colors are probably due to the limited Windows implementation of a terminal, but good enough for us now to know that something is received.

If you don't see anything with `ft-server` on the LED matrix, you need to systematically figure out what is different from running the demo. (for instance compared to the terminal version of `ft-server`, you'd run it with `sudo`; all the `--led-...` commandline options should be the same). Double check that it is compiled from the latest version (remove, fresh checkout) etc.

hzeller, 2020-03-18:
(The terminal program is the one displaying the text coming from the Pi, so if the Pi sends the color codes, the terminal that runs on your workstation needs to be able to deal with it)

jabulonboaz, 2020-03-18:
> Yeah, looks like your terminal can't deal with the color codes, but the color salad shows that something was received :)
>
> The broken colors are probably due to the limited Windows implementation of [...]

The ft server does not accept the flags 
Do u mean by --led something like this?
[image]

hzeller, 2020-03-18:
You did not compile it with `make FT_BACKEND=rgb-matrix`

jabulonboaz, 2020-03-18:
> You did not compile it with `make FT_BACKEND=rgb-matrix`

I dont follow suld i run this comand in there?

jabulonboaz, 2020-03-18:
Oh i see ok now is different 
[image]

hzeller, 2020-03-18:
to build ft-server, so that it outputs on the rgb-matrix, you have to build it with `make FT_BACKEND=rgb-matrix`.

Then, if you invoke it, it will output to the matrix.

jabulonboaz, 2020-03-18:
I vlc and nothing i dont get any video on the panel and nothing on the terminal

jabulonboaz, 2020-03-18:
Not even get the fpf now and i tried with simple example too
[image]

hzeller, 2020-03-18:
That terminal output is different is expected: The terminal will not show anything except the 'listining on 1337' or fps because now you output to the panel. To see that the ft-server is actually sending stuff to the led matrix, you can add the `--led-show-refresh` flag.

Don't give it the `-D` flag, as now the size is automatically determined from your rgb matrix.

jabulonboaz, 2020-03-18:
There is something being sent and displayed now. I changed some of the flags on ft server. The pixels are moving when vlc is streaming and stop when i pause the video 
[image]
[image]

jabulonboaz, 2020-03-18:
SUCCESS SUCCESS 
Now i need to figure out why the video file is not played back correctly maybe the VLC settings with the codec 
[image]
[image]

jabulonboaz, 2020-03-18:
The picture is shaky and there are lines in the middle maybe u can suggest why but at least now i see it :) 
[image]
[image]

hzeller, 2020-03-18:
Yay!
The lines might just be something from not enough `--led-slowdown-gpio`.

jabulonboaz, 2020-03-18:
> The picture is shaky and there are lines in the middle maybe u can suggest why but at least now i see it :)
> [image]
> [image]

Added gpio slowdown 2 now the pic is perfect.
Off to try vlc now.
We typed at the same time.
I wonder what is the correct vlc setting for an mp4

hzeller, 2020-03-18:
I just compiled a vlc and it works fine with mp4 and also a YouTube URL, sending to my flaschen-taschen LED panel.

But of course to do the mp4, it needs the right codec installed on the system.

jabulonboaz, 2020-03-18:
> 
> 
> I just compiled a vlc and it works fine with mp4 and also a YouTube URL, sending to my flaschen-taschen LED panel.
> 
> But of course to do the mp4, it needs the right codec installed on the system.

can you pls give me more deatils ? where did you instaled VLC and what settings did you use to play back youtube pls?

hzeller, 2020-03-18:
Follow the build instructions. On my Linux laptop, that is fairly straightforward, don't know on Windows.

Then I just ran it with
```
./vlc --vout flaschen --flaschen-display=192.168.1.155 --flaschen-width=160 --flaschen-height=96 "https://www.youtube.com/watch?v=yS1ibDImAYU"
```

[42 more replies]

closed 2020-03-22 by hzeller