hzeller/rpi-rgb-led-matrix #1007: Adding source of panel rails to mount RGBPanels together

marcmerlin, 2020-03-16:
It would be fantastic to have people contribute a source of rails that can be used to mount a bunch of panels together.  
For instance, I have these, and haven't used them due to lack of rails and time/skill/material to make my own:
[image]

Similarly, 
https://github.com/hzeller/rpi-rgb-led-matrix/raw/master/img/user-action-shot.jpg
is great, but would be even more awesome with a picture of the back, technical specs (resolution, how many panels, what chaining configuration, etc...)

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#chains has a great example: https://github.com/hzeller/rpi-rgb-led-matrix/raw/master/img/powerbar.jpg
but no clear information on whether this is hand built or whether one can easily buy stuch rails pre-made.  
I understand there is a length issue, but there are several ways to have shorter size rails that could be stacked to make bigger ones in a modular fashion.

Sorry if this is just wishful thinking :)

hzeller, 2020-03-16:
The first link (with the "queen") is some user built, I don't know how it looks from the back.

The second one is my first panel. It is just standard hardware from the hardware store with a lot of hand-drilled holes :) The bars in the middle are essentially some aluminum bar, the edges 1/2" L-shaped aluminum profile.

If you want modules, I think the same sellers on alibaba that sell the panels also have larger modules (the kind that ultimately make up the huge advertisement signs these are used in).

marcmerlin, 2020-03-17:
thanks. Not really looking for modules, just wondering if the community has found modular rails that already exist so that I don't have to make my own like you did :)
(it's especially a pain if you change your mind on the layout you wanted to put them in)
Sorry for using an issue for this, given that there isn't a user board/chat, I didn't find a better way.
Feel free to close this issue, not sure if leaving it open might make it visible to others who might chime in (probably a very long shot)

Beersatron, 2020-03-17:
I used [this](https://www.homedepot.com/p/Everbilt-1-3-8-in-x-36-in-Zinc-Steel-Punched-Flat-Bar-with-1-16-in-Thick-802037/204225755), the holes "nearly" lined up perfectly but all I had to do was get a metal file and reshape some of the holes. I then used machine screws and washers to fix it all together.

Beersatron, 2020-03-17:
[image]

marcmerlin, 2020-03-17:
Thank you @Beersatron this is totally what I had in mind, you rock.
Well, almost. I assume the holes don't match because imperial vs metric, but somehow with minimal work you were able to get it to fit and the panels are truly side by side without movement?
One annoying thing I found out is that they're not even perpendicular, but trapezoidal, what the hell was this for?
[image]
Do you have the same gap between your panels?

marcmerlin, 2020-03-21:
@Beersatron thanks to you I was able to build this in maybe 6H or so (longer than I wanted, but faster than making my own rails and holes)
[image]

I only had to use a very good pair of bolt cutters and was able to use them to cut the rails at the right length. Holes did not align perfectly but with careful layout and trial and error, I was able to make it work (I also had a handy collection of M3.5 screws and washers).

@hzeller I'm happy to write something up for your README.md in any directory, although I'll ask you for review on the 2 previous ones I already wrote up and sent you if you're interested in more :)

Beersatron, 2020-03-21:
> @Beersatron thanks to you I was able to build this in maybe 6H or so (longer than I wanted, but faster than making my own rails and holes)
> [image]
>
> I only had to use a very good pair of bolt cutters [...]

Awesome!

I never thought to join them vertically as well as horizontally ... that would make my setup so much easier to move and keep in shape! =D

marcmerlin, 2020-03-23:
@Beersatron  I ended up adding my shopping list in the page:
http://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html#shopping

@hzeller I'm not sure how you feel about adding recommendations for mounting hardware, which was the original point of this issue. Options are:
1) You close this issue and hope that people can find it with google and find my page with the links (which I'll paste below as a copy). This may be good enough, my problem is fixed, so I'm fine, but of course, this is also about helping others
2) you copy the relevant bits into your page
3) I send you a PR, you tell me how it's not quite what you wanted, we adjust it a few times, and then you merge it in ;-)

Shopping list:
- rails with holes mostly in usable places: https://www.homedepot.com/p/Everbilt-1-3-8-in-x-36-in-Zinc-Plated-Punch-Flat-Bar-800337/204325640
- M3 screws and washers: https://www.amazon.com/gp/product/B07FCN64HV/ref=ppx_yo_dt_b_asin_title_o05_s00
- you'll need something to cut the rails that are too long. I used a set of bolt cutters I had, some saws may work too, but the rails are pretty thick

marcmerlin, 2020-04-07:
@hzeller  would you like a doc PR , or should I just close this and hope google will do a good enough job sending people here?
I also put the links on http://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html#shopping

climy, 2021-10-20:
There is not much except for custom projects like this one:  https://www.instructables.com/O-R-A-RGB-Led-Matrix-Wall-Clock-and-More/

For my prototype, I just mitered architectural .5x.5 Aluminum angle and some extra pieces for gussets to a welded four 32 x 32 LED matrix panel frame.  Use the hot propane (MAP) gas and Aluminum filler rod from home improvement stores.  The trick is, panels will melt so fixturing it all square so panels fit after the frame cools.  Then drill holes through to match the panel mounting threads.

[image]

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin