hzeller/rpi-rgb-led-matrix #1010: How to implement this custom pixel-mapper?

punyanagari, 2020-03-18:
Dear Zenner, I am trying but can't figure out the logic and required changes to implement this custom pixel mapper. Please I need your help

I have to divide 36 panels in 3 parallel chains but visibly it will be 4 chains. (9 chains per row visibly)

So I have to change the chain after 12th and 24th panel.

Arrangement: 

[<][<][<][<][<][<][<][<][<]  }-- Pi connector #1       -----row 1
[>][>][>] Pi connector #2 -- } [>][>][>][>][>][>]     -----row 2
[<][<][<]  }-- Pi connector #3 [<][<][<][<][<][<]     -----row 3
[>][>][>][>][>][>][>][>][>]                                        -----row 4

hzeller, 2020-03-22:
You could probably use the [V-Mpaper](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#v-mapper-and-vmapperz-vertical-arrangement) that @marcmerlin just recently added.

To make it work you will have three vertical columns with a height of four 'panels'.
But wait, you say, you want them 9 wide. You can do that by pretending each triple of panels is actually one panel by giving it a `--led-cols=96` for instance (if your panels are 32 wide: 3 * 32 = 96).
So `--led-parallel=3 --led-cols=96 --led-chain=4`
Then you can use it with a `V-mapper:Z` configuration.

``` 
[>][>][>]   [>][>][>]   [>][>][>]
|           |           |
[<][<][<]   [<][<][<]   [<][<][<]
        |           |           |
[>][>][>]   [>][>][>]   [>][>][>]
|           |           |
[<][<][<]   [<][<][<]   [<][<][<]
        |           |           |
       c1          c2          c3
```

36 panels is a lot, but if they are 32x32 and you use `--led-pwm-dither-bits=1` or 2 then you might get decent refresh rate out of it.

punyanagari, 2020-03-23:
Currently I have 3 panels of 32x32 so I used the below config to test this:

`--led-multiplexing=1 --led-cols=96 --led-slowdown-gpio=2 --led-pixel-mapper=V-Mapper -D3`

[image]

`sudo ./text-example -f ../fonts/8x13.bdf --led-multiplexing=1 --led-cols=96 --led-slowdown-gpio=2 --led-pixel-mapper=V-Mapper`

[image]

marcmerlin, 2020-03-23:
@hzeller very good point, that hack about going wide and high by emulating wider panels.
@punyanagari are you sure your panels are 32x32, they don't look square at all.
Are they all cabled right to left panel pointing up?
With 3 panels, you don't even need led-pixel-mapper=V-Mapper actually, it will only be useful when you had a 2nd row of 3 panels above.
Please try -D0 with just one panel, make sure that works, and then do it again with 3 panels wide without any mapper with chain=3 and --led-cols=32, and then try again with --led-cols=96 --led-rows=32   (--led-rows=32  seemed to be missing)
Then you can add the mapper with for sanity check, but it shouldn't make a difference

punyanagari, 2020-03-23:
@marcmerlin yes panels are 32x32 they are square
Yes cabled right to left
You are correct don't need to use mapper to test this actually
The thing is --led-chain=3 is not giving the same output as --led-cols=96. And --led-rows=32 is default so haven't entered that

Output with `--led-chain=3`

[image]

Output with `--led-cols=96`

[image]

marcmerlin, 2020-03-23:
@hzeller does this make sense to you?

marcmerlin, 2020-03-23:
@punyanagari  can you try 
--led-chain=1 --led-cols=32
and 
--led-chain=1 --led-cols=64
and
--led-chain=1 --led-cols=96
to see if the squares are centered up/down in all 3 cases
Note that you should see 3 square in all 3 cases, but some will be "primary" while others will be shifted "echoes", but either way, they should be centered up/down

punyanagari, 2020-03-28:
`--led-cols=32`

[image]

`--led-cols=64`

[image]

`--led-cols=96`
[image]

marcmerlin, 2020-03-28:
Thanks. Ok, things are likely weird because you have/need --led-multiplexing=1 but in theory it should work just the same, and clearly it doesn't.
I'm not familiar with the multiplexing code, but it may be causing what you're seeing.
At this point, I'm honestly not sure, this up/down offset shouldn't happen. I'll leave it to someone else to help if they can.

arahasya, 2020-04-01:
@hzeller?

marcmerlin, 2020-04-01:
So, this probably won't help you much, but if I had to guess, the problem is that your panels require --led-multiplexing=1
I'm not too familiar with how they work, but I think you need to push pixels on a line until they get to the end, and then they go onto another line on the same panel, before exiting the panel to the output connector.
Maybe there is a bug with that + chaining where they end up on the wrong line by the time they get shifted out to the 2nd panel, and that therefore chaining will be more complicated than normal.
I've never chained panels that had internal multiplexing, so maybe there is a bug in that case? If you are able, you could dig in the code, or wait for @hzeller , but from the commit history, it seems that he's been away for 8 days now.

punyanagari, 2020-04-12:
Yes, I tried with --led-multiplexing=4 and it works, which multiplexing setting are you using?

marcmerlin, 2020-04-12:
oh, great to hear. So, now everything is aligned and working I assume, and once you add panels and my mapper, you should be ok with your array.
If so, do you mind resolving this issue?

punyanagari, 2020-04-12:
actually it's still not working for --led-multiplexing=1. I guess it's some bug?
Should I still resolve this?

Marc can you have a look at this issue please? #1035 
I need to solve this desparately

marcmerlin, 2020-04-12:
You said it worked with --led-multiplexing=4 
why are you wanting it to work with --led-multiplexing=1?
for your other bug, sorry, I don't work on this codebase, I just wrote a mapper for my own array, which I was trying to help you to use, the rest of the code, I'm not familiar with

closed 2020-04-20 by arahasya