hzeller/rpi-rgb-led-matrix #1011: 16x32 RGB Led Matrix with Scrolling Message, what I'm doing wrong ?

TristanGodal, 2020-03-18:
Hi. Sorry for disturb. 
I'm new with RasberryPi, and I'm using a Raspberry Pi 3B+.

I'm trying to use an 16x32 RGB displayer from AdaFruit, so clone this github and use the command "make" at first, then, I try to use this program :
`import os
from PIL import ImageFont
from PIL import Image
from PIL import ImageDraw
 
text = (("Raspberry Pi ", (255, 0, 0)), ("and ", (0, 255, 0)), ("Adafruit", (0, 0, 255)))
 
 
font = ImageFont.truetype("/usr/share/fonts/truetype/freefont/FreeSans.ttf", 16)
all_text = ""
for text_color_pair in text:
    t = text_color_pair[0]
    all_text = all_text + t
 
print(all_text)
width, ignore = font.getsize(all_text)
print(width)
 
 
im = Image.new("RGB", (width + 30, 16), "black")
draw = ImageDraw.Draw(im)
 
x = 0;
for text_color_pair in text:
    t = text_color_pair[0]
    c = text_color_pair[1]
    print("t=" + t + " " + str(c) + " " + str(x))
    draw.text((x, 0), t, c, font=font)
    x = x + font.getsize(t)[0]
 
im.save("test.ppm")
 
os.system("./demo --led-rows=16 --led-chain=1 -D 1 test.ppm")`

-> Code from https://learn.adafruit.com/16x32-rgb-display-with-raspberry-pi-part-2/software

But. When I execute it using 

`sudo python test.py`

The program says in console that all is ok, and that it's reading the image. 
But, nothing is currently displaying on the Led pane.

I verified the wiring, all seems to be okay, and the code from there : https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/experimental-python-code

is working very well as expected.

Maybe someone got an idea ?

Thanks. 

P.S : Sorry for my english.

raspberry-pi-maker, 2020-03-18:
I think you first need to check the path of demo executable, test.ppm files
 
In the python file,  save the test.ppm file with full path and dont call os.system().
Open a terminal, then execute this command, and check the result.
Surely you need a root permission.
./examples-api-use/demo --led-rows=16 --led-chain=1 -D 1 "test.ppm (with full path)"

hzeller, 2020-03-18:
Please don't use the demo to scroll an text. This is a super-simplistic demo that is really only used to test.

If you want to scroll text,  use the dedicated provided [text-scrolling utility](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#text-scroller)

If you _do_ want to use Python, install the  [Python library for the rgb-matrix](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python) and then use the library from within there; there are [samples](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python/samples) that show how to do that.

You call for all kinds of trouble when you mix Python and then os.system() call other programs. Do it natively either in Python or C++.

closed 2020-03-22 by hzeller