hzeller/rpi-rgb-led-matrix #1012: Is U-mapper hard-coded for 64x32 panels instead of auto-detecting?

marcmerlin, 2020-03-21:


hzeller, 2020-03-21:
it is testing if this is multiple of 32, which is the typical length of thes panels.

Having said that, it should just be able to divide by two for some odd-sized panels.

marcmerlin, 2020-03-21:
I'm talking about this. I undrestand that it's not the same as visible_width = matrix_width / 2, but while I don't fully understand the logic, I don't think it will work right on a 32x16 panels, which might be obsolete, but I still have some :)
```
    const int visible_width = (matrix_width / 64) * 32;
    const int slab_height = 2 * panel_height;   // one folded u-shape
    const int base_y = (y / slab_height) * panel_height;
    y %= slab_height;
    if (y < panel_height) {
      x += matrix_width / 2;
    } else {
      x = visible_width - x - 1;
```

hzeller, 2020-03-21:
If you have an u-shape, there are at least two panels in a chain, which means they are at least 64 pixels wide (the smallest panel I have seen is 32px long).
So this should work with 32x16.

marcmerlin, 2020-03-21:
Aaah, good call.  
But, still how is that different than writing visible_width = matrix_width / 2 ?  
Sorry if I'm not as smart as the person who wrote this, or twice as smart, depending on how clever the code, is :)

hzeller, 2020-03-21:
I think this was mostly to not have weird situations when having an odd number of 32 panels in a row. Don't remember.

marcmerlin, 2020-03-21:
yeah, that was the only thing I could see. Anyway, seems like it's as designed, sorry for bothering you :)

closed 2020-03-21 by marcmerlin