hzeller/rpi-rgb-led-matrix #1015: RPi with 16169 LED Controller

EnBee1981, 2020-03-22:
Hello,

does anyone have any experience with the IC16169 controller at a raspberry pi. My LED panel has the HUB75E connector. 

My indoor panel 

https://www.alibaba.com/product-detail/SMD-full-color-3in1-rgb-led_60839777344.html?spm=a2756.order-detail-ta-ta-b.0.0.b61f2fc2D4dii7

function is ok and the the controller is ICN2037.

The outdoor panel has the IC16169 controller. But i dont know the difference and the company dont help me.

kind regards nico

hzeller, 2020-03-22:
Did you try [through all](https://github.com/hzeller/rpi-rgb-led-matrix#types-of-displays) `--led-row-addr-type`, `--led-multiplexing` and `--led-panel-type` ?

andytheengineerguy, 2024-01-04:
Hi @EnBee1981, did you ever manage to get the IC16169 chipset working? I am facing the same problem now. I have tried all  combinations (up to the first ~5 variables) of --led-row-addr-type, --led-multiplexing and --led-panel-type and am getting no output at all with any of them (but interestingly, the panels are still drawing ~400mA of current each)

I suspect this chipset needs a new _initialization sequence_, but I can't figure out how this library does it. And even if I did, I'm not sure which variables on the datasheet I should be paying attention to! I created custom mappers in C for various panels, so I would say I'm quite capable with this library, but I'm worried this might be beyond me... 🫠

EnBee1981, 2024-01-04:
No I did not get working. If you have any solution please let me know

andytheengineerguy, 2024-01-04:
Hi @EnBee1981, an update:

I'm investigating what LED driver chips manufacturers are using in their outdoor, 320mm*160mm, P4, 80*40, 1/10 scan rate panels.

It looks like this panel from Huaruimin uses a ICN2037 chip:
[image]

AliExpress link to the panel: https://a.aliexpress.com/_m0N9WJQ

You mentioned in above comments that you successfully used a panel ICN2037 right? Could you elaborate on the settings you used?

If true, then we should be able to use these panel types without needing to write a custom initialization sequence

closed 2025-11-27 by marcmerlin