hzeller/rpi-rgb-led-matrix #1016: Rss Scrolling, im having trouble

breakingbk, 2020-03-22:
Hello Mr Zeller, 

Huge thanks for this amazing work, please may you assist me in a problem im having.

I've downloaded a python script which scrolls news from rss feeds. 

The code is here that i am using, i see you have helped on this one a few times in the past :)
https://gist.github.com/chubbyemu/4ca0c68878c6d978d067da4a36bcc71d

Only been using linux a couple of weeks so im fairly new, i did a pip install of feedparser for python.

I am able to confirm that feedparser can be imported in python, and shows it comes from '/home/pi/.local/lib/python2.7/site-packages/feedparser.pyc'> within python, im guessing that the pip install has placed it in the wrong directory or i have made some other silly error somewhere. 

I am getting an error which says File "rssparse.py", line 7, in <module>
 import feedparser
ImportError: No module named feedparser

If you could help, it would be so appreciated and thanks again.

Chris

hzeller, 2020-03-22:
I myself am more a C++ person, but there is probably someone out here that can chime in and help setting up the Python environment that it works for you.

You mentioned that you confirmed that feedparser can be imported into Python. But then you wrote a script that couldn't import it. What was different ?

breakingbk, 2020-03-22:
Hello, 

Thank you for the response, i used chmod 755 and it looks as though the script will run, it also populates the directory with ppm's which are relevant to the rss feeds i wanted 

[image]
[image]

hzeller, 2020-03-22:
do you have the right path to the demo binary ? See the error output.

breakingbk, 2020-03-22:
Do you have an idea of what would be creating this error reporting?

breakingbk, 2020-03-22:
I shall try that now

breakingbk, 2020-03-22:
is it the line 95 i should be changing?

os.system("sudo ./led-matrix -r 16 -c 3 -t 60 -m 25 -D 1" +disp) 

I have tried changing it to a the path of exaples-api-use and it didnt work 

Sorry for the noobness :)

hzeller, 2020-03-22:
The program is called `examples-api-use/demo`.

Also, you want a fully qualified absolute path (e.g. `/home/pi/rpi-rgb-led-matrix/examples-api-use/demo` ... really depending on where you checked it out)

breakingbk, 2020-03-22:
Cheers for helping me, 

im still having errors 

[image]

[image]

breakingbk, 2020-03-22:
apologies for the orientation of the photos lol :/

breakingbk, 2020-03-22:
Nothing is ever simple in IT is it :)

hzeller, 2020-03-22:
You have a dot-slash in front of your path, not a slash.
The `.` is the current directory, so you are trying to find something relative to your current directory ... and there is no home/pi/... in your directory, so of course os.system() will not find it.

If it starts with slash, this is the absolute path.

breakingbk, 2020-03-22:
Thanks, i am getting there now i think i just need to change the options now.... :)

breakingbk, 2020-03-22:
We've done it!!!! 

[image]

You are so so kind, honestly i am so happy with this project, it is absolutely amazing, if i can contribute in any way please let me know, via a donation or anything. 

Thanks so much!

breakingbk, 2020-03-22:
just need to play around with fonts and sizing and speeds etc...

hzeller, 2020-03-22:
If you just want to output text, you might consider using the [text-scrolling utility](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#text-scroller) directly instead of the detour of creating a PPM first, then run the demo.
There you have options such as background color or an outline around your font.

And the very best thing of course it to *not* invoke the external program with `os.system()` in the first place, but do everything in Python using the provided [rpi-rgb-led-matrix Python binding](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python). There are some examples that get you started with outputting text, like the [samples/runtext.py](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/runtext.py).

breakingbk, 2020-03-22:
Thanks, i shall definately have a look at doing that. Lots to learn! :) This project has really inspired me, cheers!

closed 2020-03-22 by hzeller