hzeller/rpi-rgb-led-matrix #1017: Can't figure out correct settings for my display

gingters, 2020-03-22:
Hi,

I connected one of my two 32x16 panels with the ElectroDragon board linked on the adapter page.
My panel has `P10-4S-3535-3216-JA1` printed on the back. The connectors are labeled with `HUB75D`. I ordered it as an outdoor panel, because I think I need bright panels.

I sadly can't read what's actually written on the IC's, as this is way too small: I don't have reasonable magnifying glasses and my phone cam can't zoom in enough for it to become readable.

I attached the panel to the first output, where the upper chain (the first one) should be connected.
When I run the demo, I only see wildly shuffled parts of the cube on the panel.

I used `--led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1` and I tried all 11 `--led-multiplexing` options with pretty all 3 `--led-panel-type` settings (default, FM6126A and DM6127).

What should I do now?
Do I need my own multiplexing? If yes, how can I figure that out?
Do I need to set individual pixels, create some sort of map where they actually show up and move them around accordingly? I'm a bit lost at the moment, sadly :(

hzeller, 2020-03-22:
When you first connect it, make sure to slow down gpio to not have artifacts due to that, `--led-slowdown-gpio=3` or so.

Did you level convert the GPIO outputs to 5V ? Not all panels can deal with 3.3V logic level.

Did you also try all `--led-row-addr-type` ?

Can you provide an image of how the square demo is output on your panel in the 'best' setting you get ? (`-D3` on the demo program).

What does the connector plug look like, do you have a picture of the input with the sillkscreen around (Essentially [like this image](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/hub75-other.jpg))

gingters, 2020-03-22:
Ah, awesome!

It was the missing `--led-row-addr-type`. With that set to 2 it works when I chose multiplexing setting 5.
I did not set `slowdown-gpio` so far, but will keep that in mind if I have more moving images and see artifacts. So far its fine without that setting.

closed 2020-03-22 by gingters