hzeller/rpi-rgb-led-matrix #1019: C# examples do only flash

gingters, 2020-03-22:
Hi,

The normal C++ examples (demo, clock and scrolling text) do work totally fine.
Now I want to try out the C# examples, but for some reason they do only flash and not light up continuously.

Based on the command line arguments I tested the demo and the scrolling text with, I initialize the panels with this object:

```
new RGBLedMatrixOptions {
   Rows = 16,
   Cols = 32,
   RowAddressType = 2,
   Multiplexing = 5,
   GpioSlowdown = 3,
}
```

But for some reason, all C# examples only shows the correct image for a fraction of a second, then the panels go black for pretty much exactly a second, and then the image is flashing again, and again a second of blackness.

Besides trying them on mono, I also ported the samples to .NET Core: I wrote an additional .csproj for them and compiled and ran them with the .NET Core SDK 3.1.200, and that showed the very same results as the mono one: A short flash with the correct display, and then a second of blackness.

Any idea what could be the issue with the C# samples?

My main idea is to write a small project in C# that connects to a server with SignalR. The server provides the data to show on multiple displays (two letters, and a countdown, in either red, yellow or green), and the project receives the data and drives the panels. So I can position multiple panels on different locations and show the same data everywhere synchronously.

hzeller, 2020-03-22:
I recently added some [new parameter](https://github.com/hzeller/rpi-rgb-led-matrix/commit/1ba867b0cb0ad5901b7d57e708b6fa027b698ea2) in the C++ API, maybe the c# api needs to be updated accordingly ? I remember last time I did that, the corresponding API had to be updated as well.
I'd like to forward this to @smdn @Michael-Richardson to take a look (these were doing updates to this regard recently).

gingters, 2020-03-22:
Ah. I can check whether the calls are in line with the C++ ones myself and try to update them if not.

Did not think about that they could be out of sync.
Will do tomorrow evening.

hzeller, 2020-03-22:
Alright, fixed the struct alignment. There were also a couple of other things wrong that I fixed in the process.

commit daf82789 (2020-03-22): Fix C-struct mapping to C# binding

  * The bitfield in the C-struct was not properly mapped. Simply
    make this a char's to reflect the boolean there and map as byte
    in C#
  * Fix the way the booleans were translated to the struct (the
    inverse value and show refresh were always set due to inverted
    ?: logic; possibly this was an previous attempt to 'fix' the
    issue with the bitfield mapping ?
  * Add missing new values for "panel_type" and "limit_refresh_rate_hz".
  * Make sure the --led-* command-line flags are passed to the library
    so that C# programs can make use of the [... cut]
closed 2020-03-22 by hzeller