hzeller/rpi-rgb-led-matrix #1020: P10 RGB Panel: Only alternate rows lighting up

jawedsoft, 2020-03-23:
Hello Everyone.
Wishing all of you are safe and taking good care of yourselves.

I am writing form from India.

I have a p10 16x32 RGB smd panel.  I have hooked it to the 40pin rPi and am running the library downloaded from git around 10 days back.

The pics of the panel are attached.

The problem I faced was that alternative rows are permanently off. All the demo are working fine but only 8 rows lit up.
 When I disconnect the header from rPi and touch it with my finger, all the LEDs light up randomly so  I am hoping that nothing is wrong with the panel. (my working config is led-rows=16, led-multiplexing=11)

I have read through the issue list on Git and came across another user who had same problem and the solution given to him was that probably the 'A' line is not getting the signal.
I have gone back and checked all my connections and find that the lines are fine. (A, B and C, all three are fine)

Can someone one please guide me where else I should be looking at.

regards,
Jawed 
[image]
[image]
[image]
[image]
[image]

hzeller, 2020-03-24:
Maybe simply the panel is broken, or did you test it with multiple of the same kind ?

jawedsoft, 2020-03-28:
ok..my issue has been resolved. 
Thank you hzeller.
It was indeed a problem with the A line not reaching the panel. The ribbon cable was broken. 
Connected the 3M cables directly one-by-one to the panel from the rPI and it worked like magic.

Thank you everyone and praying for safety from n-Cov.

regards,
Jawed Ahmed India

closed 2020-03-28 by jawedsoft