hzeller/rpi-rgb-led-matrix #1022: Different points in hub75 64x32 rgb led

sahukanishka, 2020-03-23:
I have a 64x32 outdoor RGB led pannels and it has a hub75 but the details of the pins are not matching with the library wiring methods. 
-----here is my led panels hub looks like 
[image]

I already connected with the same pattern you have shown in the wiring guide and it is showing the wrong output of the scrolling text ---
[image]
 kindly, please help 
Thank you

punyanagari, 2020-03-23:
did you try --led-slowdown-gpio=3?

sahukanishka, 2020-03-23:
I'm totally noob in this field. 
No i did't try this .
Thanks for replying i'll give try.

closed 2020-03-27 by sahukanishka