hzeller/rpi-rgb-led-matrix #1024: C# Binding: font-example breaks, probably because of cmdline args (font path and text)

gingters, 2020-03-23:
Hi,

first of all, thanks for fixing up the bindings so fast yesterday!
I got all C# binding examples working out of the box with *both* Mono as well as .NET Core, except for the `font-example`. This does work in my .NET Core version, but not in the default on Mono.

The problem with the font example in Mono seems to be in the newly changed .ctor of the `RGBLedMatrix`, where it gets `Environment.GetCommandLineArgs()` and passes that along as `argv` to the native call `led_matrix_create_from_options` where that should get parsed.

In the font example, the first argument should be the path to the font, and the second the text to display, and somehow the P/invoke call then causes an exception when the `led_matrix_create_from_options` is called with this command line string.

In .NET Core, for whatever reason, the call works. 🤷‍♂️

hzeller, 2020-03-24:
I couldn't reproduce this on my pi, but I can imagine there might be issues when the array is modified by the c-function; this sounds like something that can get a managed/unmanaged transition into trouble.

Given that we don't actually care changing the value back, I just added a version of the  `led_matrix_create_from_options` that does not modify the array. Does that work for you now ?

hzeller, 2020-03-24:
mmh, so the interesting thing is, that example does not even invoke the constructor that takes options, but the [simple one that just take three parameters](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/c%23/examples/font-example.cs#L21).

So something else is going on in your case.

gingters, 2020-03-24:
I changed the call to use the ctor with the options, so that I can pass my addr type, rows and multiplexing in. And with that I got the exception. But if that does not happen on your side it may be a problem with my Pi and my Mono setup.

That said, would you be open for me doing some work on the C# binding?

I'm only afraid I am going to propose a few changes that might be breaking, especially changing the namespace (you really don't see lowercase namespaces with underscores in the .NET world).
What I could do is to additionally keep the old namespace with a compiler directive for backwards compatibility and do only a type-forwarding to the new ones.

Then I would only use the command line arguments from the environment if the developer does not pass along along a (potentially pre-parsed) argument array and some general cleanups like one file per class/type, as it is very common in C# projects. Also, the options object could be pre-initialized with the defaults and the settings could be exposed as enums. This of course would bind them tighter to changes on the C++ side, but its more convenient for the C# developer to use.

On top I would provide additional .csproj files to be able to build a .NET Core classlib out of the binding (so that you can select either Mono or .NET Core by default). As of now I just don't know if I can call the .NET Core compiler from Make (I don't know make at all, but this is a possibility to learn).

I would also like to extend the samples and port especially the scrolling text to C# (as I need that for my own project anyways), and possibly also the other samples that are missing as of now.

What do you think?

hrocha1, 2020-05-17:
Having examples for .NET Core in the main repository would be amazing. The library that @gingters made seems to be really well documented and easy to use.

gingters, 2020-05-17:
Thanks.

My port however is patched in a few places to get access to other API calls.

And it doesn't work with Pi Zeroes as .NET Core does not support that ARM version.

Also, I'm a bit opinionated on how .NET Core code should look like ;)

Holen Sie sich Outlook für iOS<https://aka.ms/o0ukef>