hzeller/rpi-rgb-led-matrix #1027: Layering and transparency?

mbates14, 2020-03-24:
I would like to have a background image, with text on top of it, as well as other images, etc layered on top of one another with transparency. 

Is there a way to layer canvases? or is there a different way to do this? 

Also what about adding Alpha to RGB? for example I have images that have gradients and other things that use Alpha (like PNGs) and the background layer would show through. 

its not as easy as creating a flat image with all the effects embedded, because the content is dynamic. Plus it would be cool to have a text fade effect when changing text. 

any ideas? Thanks!

hzeller, 2020-03-24:
The canvas provided by the matrix only provides the 'raw' output.

But it should be fairly simple to implement a delegating canvas that does the compositing and then forward the resulting thing to the underlying canvas. Implement something that implements the Canvas interface and holds the raw RGB values in a buffer, so that you can modify them as needed. Then you delegate to some underlying Canvas which you give the resulting RGB.

mbates14, 2020-03-24:
ok, but I have no idea how to implement any of that, so what does that look like? 

as far as I saw, I only saw a setpixel so im not sure how to do anything like that?

hzeller, 2020-03-24:
you can just store a raw byte-array or maybe an unsigned short array to avoid rounding issues.

Provide a `SetPixel()` with an additional alpha value. Use that to calculate the new value including alpha in your internal buffer and then call the call the underlying `SetPixel()` of the canvas you are delegating to with the new color. You have to keep the array internally so that you can update things according to the alpha value.

To implement `virtual SetPixel()`, it will just delegate to the SetPixel() with alpha.

mbates14, 2020-03-25:
Does setpixel allow Alpha?

hzeller, 2020-03-25:
No, you have to implement that yourself. For something like alpha, you need to have access to the existing color at that point, this is why you have to implement a framebuffer where you can look that up.

mbates14, 2020-03-25:
ok. I think I am more confused than when I started :-(

hzeller, 2020-03-25:
What is the confusing part ?
Implementing Alpha channels always means you have to mix with the previous content of your image. So you need to keep track of what you send to the Canvas to be able to do the mixing.

mbates14, 2020-03-25:
Well I have never low-level coded alpha before. I was hoping the library supported it.

hzeller, 2020-03-25:
It is not hard, it just a matter of implementing it :) Since you're the first one needing it, maybe you can get things organized that it gets implemented and put in a pull request.

mbates14, 2020-03-25:
Says the guy that has a super in-depth knowledge of how all this works. 

I am green, I need someone to hold my hand. haha.

closed 2020-03-25 by mbates14