hzeller/rpi-rgb-led-matrix #1028: Question, what is the easiest way to display an RSS news feed with text-scroller ?

jabulonboaz, 2020-03-26:
Again thank you for this amazing got Hzeller and for all of the support you provide truly remarkable work.

Question, what is the easiest way to display an RSS news feed with text-scroller ?

sachasayan, 2020-03-29:
That's a very broad question. The implementation provided is to display text, not to parse RSS — that's up to you. 

You might want to consider looking into @alexeden's Node/TS bindings for this project, which will allow you access to a greater number of internet-related APIs and functions out of the box give its Node base: 

https://github.com/alexeden/rpi-led-matrix

Cheers

jawedsoft, 2020-03-30:
I am using runtext.py and its works properly. My code collects data from feedparse and populates a text string. The string is cycled through the scroller and at fixed intervals the feedparse updates the string.
I am still tinkering with it. But its works properly.
regards,
Jawed

jabulonboaz, 2020-03-31:
> I am using runtext.py and its works properly. My code collects data from feedparse and populates a text string. The string is cycled through the scroller and at fixed intervals the feedparse updates the string.
> I am still tinkering with it. But its works properly.
> regards,
> Jawed

Nice to hear sounds exciting, please share your code and some instructions here so that I can try that too. Thanks

jabulonboaz, 2020-03-31:
> That's a very broad question. The implementation provided is to display text, not to parse RSS — that's up to you.
>
> You might want to consider looking into @alexeden's Node/TS bindings for this [...]
 This is too generic with no reference to RSS I don’t se how this would help me.

jawedsoft, 2020-03-31:
[jawed-scrolling-news.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/4407257/jawed-scrolling-news.txt)
Hello Everyone,
I  have shared my code example. Please don't judge my coding skills. I have taken code from different places and stitched together the snippets to show me the news. I am not a coder but I am willing to learn and improve.

Some of the things which I could not do in my code are:
1. Change the panel brightness at run-time so that the panel is bright at day and dims in the evenings
2. Show data from multiple RSS feeds
3. Start automatically when the Pi reboots..

Gradually, I plan on incorporating them.

Take care and be safe

regards,
Jawed Ahmed, India
Video Link : https://youtu.be/jO3w1CLbEFs
[image]

jabulonboaz, 2020-03-31:
> [jawed-scrolling-news.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/4407257/jawed-scrolling-news.txt)
> Hello Everyone,
> I have shared my code example. Please don't judge my coding skills. I [...]

Thanks for this ! I will try it and then report back.
I am not a developer ether so thanks again.

jabulonboaz, 2020-04-03:
@hzeller is there an easy way to display an RSS feed say google news headlines using the C baste test scroller — parameters rather than writing new code pls?

jabulonboaz, 2020-04-03:
> [jawed-scrolling-news.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/4407257/jawed-scrolling-news.txt)
> Hello Everyone,
> I have shared my code example. Please don't judge my coding skills. I [...]

Hello where in samplebase.py did you input your options pls? Did you edit the default values of the code to match you panel’s characteristics or how exactly pls? I was looking at the file and not sure how to edit it so it can take my panel parameters in to consideration.

Jani73, 2020-12-26:
Here’s one that uses hzellers code.

https://youtu.be/omMVAtGGr_0
Read the description
It creates images from rss-feeds and scroll those.

SnowyDevelop, 2022-01-20:
> Hello where in samplebase.py did you input your options pls? Did you edit the default values of the code to match you panel’s characteristics or how exactly pls? I was looking at the file and not [...]

I wanted to use that but every time i want to use it this happens!
[image]

SnowyDevelop, 2022-01-20:
> [jawed-scrolling-news.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/4407257/jawed-scrolling-news.txt) Hello Everyone, I have shared my code example. Please don't judge my coding skills. I [...]

I wanted to use your very nice code but when i try to execute it this happends:
[image]

closed 2025-11-27 by marcmerlin