hzeller/rpi-rgb-led-matrix #1030: limit refresh rate missing from Python?

mbates14, 2020-03-26:
I am trying to keep the refresh limit to 120hz and lower so itll be more compatible with cameras aimed at the display. 

However, when adding that option, it giving me this error:
 sudo ./TNCApp.py -p /dev/ttyUSB0
Traceback (most recent call last):
  File "./TNCApp.py", line 231, in <module>
    options.limit_refresh_rate_hz = 120 #Limit refresh rate to 120hz, Makes it more compatible with cameras.
AttributeError: 'rgbmatrix.core.RGBMatrixOptions' object has no attribute 'limit_refresh_rate_hz'

thoughts?

maartenlambrecht, 2020-04-29:
I noticed the same thing.
there does not seem to be any setting "--led-limit-refresh=<Hz>" in Python.
I could set every other setting but that one.

```
options = RGBMatrixOptions()
options.rows = 64
options.cols = 64
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'regular'
options.row_address_type = 0
options.multiplexing = 0
options.pwm_bits = 11
options.brightness = 50
options.pwm_lsb_nanoseconds = 135
options.led_rgb_sequence = 'RGB'
options.pixel_mapper_config = ''
options.gpio_slowdown = 3
options.show_refresh_rate = 0
options.disable_hardware_pulsing = False
```

inv3rtedco5ine, 2020-05-31:
This was resolved with the pull request above

mbates14, 2020-07-16:
I verified that it works. Thanks.

closed 2020-07-16 by mbates14