hzeller/rpi-rgb-led-matrix #1035: Converting from uint_32 to uint_64 for Compute Module Chain 5 and 6

punyanagari, 2020-03-28:
I am trying to make changes from errors in compiling but not able to solve the problem.

https://github.com/punyanagari/rpi-rgb-led-matrix

--led-gpio-mapping="compute-module"

punyanagari, 2020-04-12:
I am getting closer to solve this.
I don't have any error now but the output is not as expected
I connected 1st RGB on GPIO 7,8,9 and 2nd RGB on 32,33,34

Here is the command:

` sudo ./demo -D3 --led-multiplexing=4 --led-rows=16 --led-gpio-mapping=compute-module`

Image: First half is correct next half is wrong showing blue colour only
[image]

What else do I need to change to uint64_t

Please advice @hzeller

salazarhog, 2020-04-15:
Closing this issue now
Successful in getting 6 parallel chains from Compute Module 3 +
Also automated the Initialization of GPIO's > 31 so that it won't affect the refresh rate when they are not used

Let me know if I should send a pull request

punyanagari, 2020-04-15:
Closing this issue now
Successful in getting 6 parallel chains from Compute Module 3 +
Also automated the Initialization of GPIO's > 31 so that it won't affect the refresh rate when they are not used

Let me know if I should send a pull request

closed 2020-04-15 by punyanagari

hzeller, 2020-04-16:
Yes, please send a pull request!

punyanagari, 2020-04-16:
Done