hzeller/rpi-rgb-led-matrix #1037: Putting level shifters near the hub75 input instead?

uwuforever, 2020-03-30:
If I were to create my own PCB with 74ahct245 that would be directly soldered to the hub75 input, with power and ground soldered to the panel's power connection, and ran a cable from each of these PCBs to the appropriate GPIO pin... would that fix the issues related with having a long cable? Or other words, by the suggestion to shorten long cables, is that just referring to the cable between the level shifter & the hub75 input, and would it help if we eliminated that cable entirely and instead replaced it with a cable between the pi and the level shifter? (my assumption was that the buffering counteracts it)

If so, in designing this new board, besides for gpio pin 18 for PWM, are there other pin assignments that have a technical significance, or are the rest ok to change? 

Last - why is a physical pull-up resistor needed; why not use the built-in one?

daveythacher, 2020-04-14:
Most of the HUB75 boards have one built in. These buffer the driver strength coming out of the Pi. If a longer cable is needed you will need to apply another buffer along the way. Shorting the cable or lowering the clock speed will improve performance. 

The nice thing about these is they are mostly single direction. Which for the most part mean that the drift should be consistent, however this is heading outside my area a little bit. Point being is drift is less of a factor than hold and setup times.

You could also replace with expensive drivers and use differential signaling. I would recommend closing this out as won't fix. Looking through reference pin layouts and GPIO module I think this is correct, but I am not sure.

daveythacher, 2020-04-15:
There is a passive version of the boards. You will want a level shifter, I check your matrixes before removing them. The pull up is there to disable the display till its time drive the display. The shift registers could be initialized with random values. The pin may not startup in the correct mode. The driver may not support a strong pull up or use open drain driver. (Probably just there as a precaution.)

The special pins are the PWM pins, which I think are 12 and 18. However the code also supports a timer based pulser implementation. It uses the PWM to step away from waiting for the OE pulse hold time. Which can be around 200ns, This can avoid a loss of performance, depending on setup.

Other than that I think you can just remap the pins in the code. This library is pretty much pure user space bit banging. Note I think you lose audio if you use the PWM pulser, since the Pi does PWM audio. (Which consumes some CPU, but its cheap!)

hzeller, 2020-04-16:
You definitely want the level shifter close to the Pi, not the Panel. The outputs of the Pi are directly wired to the internals of the processor with no buffering in-between. You certainly don't want to drive long cables with it.

closed 2020-04-16 by hzeller

daveythacher, 2020-04-16:
There is buffers in the chip. They are just something like CMOS buffers or weak buffers, like open drain or NMOS/PMOS. These buffers are not power transistors, however they do have output buffers. Most of the time the core logic uses a lower voltage than the output voltage. These buffers boost this and decouple other things internally like fan out performance. (However that's getting into chip internals.) 

These are for stuff you do to work with logic circuits close to chip on PCB, and some support for fan outs. They are capable of driving short wires directly, however as mentioned long wires would be a problem. However you can get into trouble if you over drive. For the speeds of this project that is not likely to be a problem. That gets into areas that are a little outside my area.