hzeller/rpi-rgb-led-matrix #1038: FPGA board?

uwuforever, 2020-03-30:
Is there plans to build FPGA support into this? I imagine it would be cool if there was an 'active-fpga' board, and we can pass a -fpga=1 option and free up a bunch of gpio pins, fix timing issues/increase refresh rate etc and have everything else work as is.

I found this: https://github.com/rjrouquette/rgb_matrix_udp but wasn't sure how to get it to work.

davemaster, 2020-03-30:
fpga would be for refreshing rgb panels arrangement with video memory content  from raspberry pi, but; what interface between rpi and fpga will be?, how much extern memory will be needed for the fpga refreshing board? Interest project....

hzeller, 2020-03-31:
I have plans in doing so, but no ETA. There are people using FPGAs already, so this might be useful.
I imagine the 'coding part', such as arrangement of pixels, multiplexing etc. in the base code, and the criticial path execution in the FPGA.

uwuforever, 2020-03-31:
I imagined the Pi sending completed frame segments to the FPGA; the FPGA would maintain a frame buffer of the latest received frame segments & code necessary to output to hub75.

Not sure if this is a dumb question but could we use a Teensy/Arduino Atmega chip as a temporary alternative? I'm assuming the main reason Teensy on Smartmatrix can only handle very small setups is due to the frame generation eating resources.

davemaster, 2020-03-31:
Could be any one of them sending frambuffer from PI by spi I think, and
receiver update the memory with current pi screen content to show in rgb
led panels arrangement

uwuforever, 2020-04-04:
Alternatively is it potentially possible to use something like tlc5958? not the greatest with EE, but it seems like it should be possible to use that chip to offset a lot of processing work easily.

uwuforever, 2020-04-06:
Re: fpga

Possible fpga board design could be a PCB with this module: https://www.xilinx.com/products/boards-and-kits/1-12hrhst.html soldered by its gpio pins, add level shifter ICs on the reverse side and 4 hub75 connectors on the edges. Power connected to the 5v from the panels.

RPi connected to FPGA via SPI/UART. (possible to have this wireless? such as: connecting the FPGA to an esp32, and using a smartphone/wireless Pi to send new frames)

Then for its software, this looks like a solid base: https://github.com/DuinoPilot/rgbmatrix-fpga

I'm guessing it would need updates to make use of the newer fpga version to support 24+ bit color, to support more than 1 chain, & for frame update protocol.

davemaster, 2020-04-06:
> 
> 
> Re: fpga
> 
> Possible fpga board design could be a PCB with this module: https://www.xilinx.com/products/boards-and-kits/1-12hrhst.html soldered by its gpio pins, add level shifter ICs on the reverse side and 4 hub75 connectors on the edges. Power connected to the 5v from the panels.
> 
> RPi connected to FPGA via SPI/UART. (possible to have this wireless? such as: connecting the FPGA to an esp32, and using a smartphone/wireless Pi to send new frames)
> 
> Then for its software, this looks like a solid base: https://github.com/Sl-Alex/rgbmatrix-fpga-fm6126a
> 
> I'm guessing it would need updates to make use of the newer fpga version to support 24+ bit color, to support more than 1 chain, to support D&E address lines & for frame update protocol.

Greetings,

Looks like a teensy that fpga board.

I have already a XC6SLX16 board with build in DDR3 interface, I will try with this one. Any advice, welcome. What interface must be used for transfer the framebuffer to FPGA ??? UART or SPI?

uwuforever, 2020-04-06:
I'm not sure how significant the difference between your Spartan 6 and my proposed Spartan 7 is, but I remember reading due to block RAM size limitation Spartan 6 colors would be 18 bit while the block size on Spartan 7 is 36 bit. I'm assuming any other limitations are for similar reasons since Spartan 7 is fairly recent. 

I think SPI is a better choice because it's faster?

davemaster, 2020-04-07:
> I'm not sure how significant the difference between your Spartan 6 and my proposed Spartan 7 is, but I remember reading due to block RAM size limitation Spartan 6 colors would be 18 bit while the [...]

totally agree about the power of Spartan-7 over 6, but it's all I have

first attemp: try spi connection between RPi 4 and xc6slx16... 
we'll see

punyanagari, 2020-04-11:
Hi can anyone explain to a noob like me the project mentioned in the first link

I went through the project. Is that micro controller acting as an FPGA here? How much will it improve the performance?

daveythacher, 2020-04-14:
This is a big topic. FPGA designs are usually application specific. Will this project likely use offloading...doubtful based on looking through the code. Given the statements made about performance and DMA, I would expect such a request to be completely outside this arena.

I suspect the color depth is a lot lower than advertised. (I was a little shocked to see that section of code.) Also the advertise chain lengths cannot support this as most of these have 25MHz max. The bit banging logic does not appear to suggest TLC5958 like drivers which macroblock does make. (TLC5958, would be an interesting one for the Pi, but an interface microcontroller could fix that. However the TLC5958 is a part that would go inside the HUB75 matrix.)

Bit banging and CPU usage is too low to suggest decent bit depth. I am not sure this ARM microprocessor supports NEON or not, however the issue there is memory bandwidth, timing issues and potential uses of user space processes. You have to look at a few things to see that.

The Pi is limited in terms of interfaces. However this can be worked around. SPI is a good one, but I would do things a little different with it. The dual buffer is a good approach for preventing tearing, however with dual port memory you could avoid it. Control interface for preventing tearing would be a pain.

The idea behind the TLC5958 is basically this. Memory locality and PWM offloading. This lowers the bandwidth that would need to move across the bus. Enabling longer chains and lowering controller cost/complexity. TLC5958 has a few imperfections in it. It has a few cool spots but it punted in a few areas.

The HUB75 boards usually have a 5 or 10 ns hold on the OE for the same reason the TLC5958 has external logic clock the PWM counters and toggle the PMOS rows. PMOS transistors have capacitance which means the row switch time is not exactly known. This is an area that ends up being a problem for FPGA designs too.

This complicated the control interface for the TLC5958. Granted this could be worked around potentially. However it complicates DMA transfers for PWM data. You see if you using the 16 bit PWM in 8 bit PWM, which you can do if you rest the counters via command after it does one pass of all the rows. (It uses E-PWM or S-PWM which means it does 256/512 steps per channel then moves to the next row. It scans the total rows (up to 32 rows) up to 256/128 times per frame. PWM clock can be up to 33MHz. Documentation is a little poor about this.)

The other mistake is it does not have copy function. Meaning you likely have to draw a whole frame into the next buffer or keep track of it. However you could just copy the darn thing internally. Dual port memory is more expensive so more than likely they optimized and use two single port 24kbit memories.

The TLC5958 does appear to be HUB75 compatible, which was a questionable call. The good news is TLC5958 has support for 3.3 volts with a software current gain logic for the constant current drivers. This would allow color correction in software for switching power from 5 volts to 3.3 volts. Most of these panels do not support that. It also has control for global brightness.

Hardware will always win. It is the law, if you say otherwise you know nothing. Perhaps you have still to learn. Closer inspection of software even shows it. As for your economics, well that's not doing so hot right now. 

I am working on a project that plays around with some of this. I have gone through several iterations which most I never finished. (Some were pretty darn silly.) The closest thing to this currently would be a project based on Teensy. However I am a little critical of that one too. (Abused the stuff out of templates.)

My approach will attempt to blend hardware and software. To utilize the strengths of each.

daveythacher, 2020-04-14:
You could convert the HUB75 into a single SPI chain. The HUB75 that works off simple constant current drivers is nothing but six parallel SPI chains. So you could link them. For QSPI you would need to pad two chains with dummy shift registers. The issue there for PWM ends up being generating the SPI frames. This could end up needing 0.75-1.2 GHz for a full HUB75 connector at full performance. However for hardware it would be maybe 0.05 GHz. Remember power consumption scales with frequency, can be linear or exponential. The LEDs will almost always use more.

daveythacher, 2020-04-14:
> I imagined the Pi sending completed frame segments to the FPGA; the FPGA would maintain a frame buffer of the latest received frame segments & code necessary to output to hub75. The buffer would just be the size of 1 complete frame, and 2 threads: 1 to update the buffer and 1 to reload the panel.
> Not sure if this is a dumb question but could we use a Teensy/Arduino Atmega chip as a temporary alternative? I'm assuming the main reason Teensy on Smartmatrix can only handle very small setups is due to the frame generation eating resources.

You are correct the Teensy is not likely able to handle large setups. However it may do fine depending on the PWM settings. The RAM could become a problem. I do not believe this project supports large PWM settings. The Teensy clock would likely get in the way of large PWM.

SmartMatrix is based on Teensy which is more powerful than XMega used in the original link. That works off timer, DMA, interrupt controller, and DMA to GPIO bank. Its the weird C++ template project. As I recall it uses more CPU than RAM. If it used large RAM it could just use DMA to draw. This would limit the video frame rate but the refresh image frame rate could be decent. I am not sure what the interface options are for SmartMatrix.

Please note I am not super aware of SmartMatrix, so please take this with a grain of salt! I am not an expert on this project.

daveythacher, 2020-04-14:
> I have already a XC6SLX16 board with build in DDR3 interface, I will try with this one. Any advice, welcome. What interface must be used for transfer the framebuffer to FPGA ??? UART or SPI?

DDR3 can be deceptive. Probably will work but the latency and bandwidth may want to be factored in. Bulk transfers may be a good idea ran into some kind of on chip memory.

daveythacher, 2020-04-14:
> I'm not sure how significant the difference between your Spartan 6 and my proposed Spartan 7 is, but I remember reading due to block RAM size limitation Spartan 6 colors would be 18 bit while the [...]

Spartan 7 is nice for soft cores. (Vivado when its not having a ton of issue makes the S7 fun. I have found issues everywhere with Vitis. PentaLinux is joke, good concept pain to work with.) However for pure hardware their may be an argument for Spartan 6's smaller blocks. Its compatible like with a large number of other FPGAs. Max10, MachXO2/3, etc.

daveythacher, 2020-08-07:
FPGA logic will need a large RAM inside FPGA for PWM steps over 256 (8 bit per color). However for steps at or below 256 you can use a small RAM in FPGA. I would recommend using several RAM modules to make the parallel data bus of the RAM match the operand, something like 48 bits for single HUB75. This will reduce the frequency to help meet timing and cut costs for high speed grade parts.

The issue with PWM on the HUB75 is bandwidth and locality, which is why those PWM and Memory mapped drivers work so well. However we can implement part of these with a FPGA for cheap. The issue with those and some of the FPGA is the timing that the interface implies. Not an issue for microcontrollers.

FPGA cost jumps for high RAM. External RAM is cheaper but requires lots of pins which can add cost. MCU in 32 bit can add RAM if you are using the approach I am about to mention. However depending on the timing model you may be able to directly drive from Linux.

The panels are nothing but a bunch of rows with column data. The PWM step is basically a loop. So you can send the data once and store it. Giving a high bang for buck. This lowers the bandwidth for talking to FPGA. Allows utilization of HUB75 bandwidth. However it would work up to 256 steps before the refresh performance suffers. Also the CPU does not have to generate bit values which is a significant effort for software.

What you can do is store two rows of column data inside a RAM. The RAM will need to be dual port or pseudo dual port. Half will be read will the other have is written by the external processor. All frame generation takes place by the external processor.

Commands will be needed to sync state, configure, transfer data, etc. I think 64 bit fix protocol should work for chain lengths up to 128 at 8 bits per color for a single chain. This will allow at least 64 commands with 58 bits reserved for command data.

You can use Lattice MachoX2 1280 QFN-32 for this. It has enough EBR to make the RAM cell using dual port memory. A small in LUT FIFO is possible to convert SPI to command processor state machine. This would write to RAM and manage HUB75 state machine.

Nice thing about this part is it ships with I2C enabled programming, which would be used to update the bitstream.  I would use SPI on the JTAG pins with JTAGENB function enabled. This would allow bit banging the JTAG to enable restoring the I2C or programming via JTAG directly. Meaning the FPGA would expose 7 pins (JTAGEN, TDO, TDI, TCK, TMS, SCL, SDA). This means no programming is require for the hardware board. Libraries already exist for accessing the I2C.

All the control wires could be generated by the FPGA logic or controlled by the commands from the external processor. There are enough pins to enable up to 5 address lines. This would allow you to support wire multiplexing schemes. All data formatting takes place in the external processor.

There could be an interrupt which signals the end of the HUB75 state machine. Allowing the external processor block if it finished data transfer of next line before HUB75 state machine. However it may be possible to just delay. This is the concern area for Linux, as time control may drift. However the CPU usage will be lower, as the underlying system will likely use DMA to transfer to SPI. In theory a delay could exist with memory stalls. Another issue avoided with dedicated memory in FPGA.

Back to how the E-PWM/S-PWM stuff works. You scan the each row for half the bit count. So 8 bit/256 step for 16 bit color. Then wrap around to the first row then increase a global counter of the upper half. In order to do this you need full imagine in memory, which causes problems. 25 MHz across shift registers is not enough for decent frame rates even at 1:8 multiplexing. The PWM drivers avoid this. Memory mapped drivers allow full scan, but have weird interfaces.

Overall this will yield better PWM depth and performance. This is a cost optimized version. A lot of the real time issues can be solved with microcontroller. Moving to higher PWM steps and display density would involve moving to panels which use the PWM and Memory mapped LED drivers. Note this will require a different version of FPGA logic, which the I2C logic could allow swapping it out. Flash write endurance is 100k.

SPI is used here so multiple FPGAs could be used. Multiple chains could exist per FPGA using different parts. Depending on SPI bandwidth. SPI bandwidth is limiting factor, which can be resolved using Zynq like systems or ASIC. 

Currently I recommend the highest speed grade, which is speed grade 6 for this part. Which increases the cost from just under 5 dollars to about 6 dollars in single unit quantities. Supporting the full memory would cost more.

In summary a cheap module could be made which address many of the issues here. I have attempted to explain the motivations. You can increase chain length with reduction of PWM step. You can also reduce the HUB75 clock. This is mostly still a work in progress.

daveythacher, 2020-08-07:
You could even make this work for steps over 256, it means more memory or cycle usage by the external processor. However you still have the drift time issue. However you do have flexibility. This overall allows you to relocate the cycle usage of PWM. Bandwidth is still an issue and there is little locality here. There is some, but not as much as the memory mapped interfaces. Bandwidth exists with the HUB75 and SPI.

daveythacher, 2020-08-17:
FTDI FT232H, FT2232H, FT600, FT601, etc. would be decent picks for interfacing with higher panel size configurations. Not aware of many boards with external memory, large internal ram, and a fully connected FTDI chip. 

You could use Arty boards with Microblaze soft core to enable a few channels. The UART connection could enable some basic support. (I would not use UARTlite, if planning to use full FTDI UART speed.) Using series 7 and Vivado you can step away from a lot of HDL coding. You can use C/C++ to write the control logic. If you generalize it correctly it will be in full control while hardware handle heavy loads in parallel. This is less efficient and costly, but there are boards and IP you can get for this.

See other posts in #918 

@Cellgalvano Here is more FPGA discussions.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin