hzeller/rpi-rgb-led-matrix #1044: Groups of Black Rows on Panels

RoastedPecans, 2020-04-08:
Hi! I've worked with this library before and gotten it to work successfully, but I seem to be running into issues replicating what I've done in the past and I'm not sure why. I'm hoping I'm just missing something obvious :)

Anyways, here is what I'm currently working with:

- Raspberry Pi 4 Model B 4GB running Raspbian
- 4 64x64 RGB LED Panels (specifically this [model](https://www.aliexpress.com/item/32733178058.html)) arranged as a 2x2 matrix
- [Adafruit Bonnet](https://www.adafruit.com/product/3211) with solder applied to short pin 8
- A Power Supply outputting 5V/4.4A to each LED panel + my Bonnet

When I try to run demo 10 from /examples-api-use using the command:

`sudo ./demo -D 10 --led-rows=64 --led-cols=64 --led-chain=4 --led-slowdown-gpio=4 --led-gpio-mapping=adafruit-hat --led-pixel-mapper="U-Mapper"`

I get the correct size (128x128), but my output comes off very strange:
[image]

As you can see, every group of 2 rows alternates between on/off. The one row with two pixels out in the middle are actually dead-pixels, not an eccentricity. I'm not sure why this is happening.

Here's the kicker: I've got this exact same set-up (down to the cables) to work before with no issues in November. So I'm not sure what's changed unless I'm missing something obvious, my panels/cables broke, or getting it to work the first time was a complete fluke.

Here's what I've tried to fix the problem:
1. Using just one 64x64 LED panel instead of all 4 (along with the appropriate command to run the demo).
2. Re-installing/Updating this library.
3. Measuring output from our power supply to ensure the panels + Bonnet are getting enough volts/amps.
4. Trying to run different demos besides 10/using the image viewer. 
5. Trying a myriad of different command line options when running the demos including different chain lengths, led-rows, led-cols, led-gpio-mappings, and more. The command given before my image is the best result I've managed to get.
6. Running [AdaFruits' script](https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rgb-matrix.sh) to install packages necessary for operation of the Bonnet.
7. Disabling Sound + Running only from CLI interface of my Pi.
8. Entirely resetting my Raspberry Pi + trying to get everything to work again from scratch. 
9. Ensuring One-Wire interface is disabled.

No matter what I try, I end up with a similar output where every group of 2 rows alternates between on/off.

As I've said before, this exact same set-up has worked before so I'm not sure what I'm missing. In the past, I didn't have to change any of my row addresses/wirings/GPIO mappings either I believe.

Does anyone have any ideas what could be causing this? Right now I'm thinking it's likely a hardware/power issue, but I'm not sure how to troubleshoot that beyond measuring the power output from my power supply. I've seen from other issues on this repo (#78, #545) that some people had issues with connections that could be causing this as well, but since it's worked in the past I'm not sure why this would be an issue now. Hoping someone might be able to point me in the right direction + double check what I've done so far to make sure I'm not missing something obvious 😄

Thanks!

hzeller, 2020-04-08:
If it is a 64x64 with E line: did you connect it https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet ?
If it is another type of 64x64 panel: did you try the relevant combinations ? https://github.com/hzeller/rpi-rgb-led-matrix#types-of-displays

RoastedPecans, 2020-04-08:
I'll try out all the combinations for command line options for different types of displays later tonight. Thanks for that link, I'm not sure what lines my panels have so maybe they're just A/B which could be causing the problems. 

If that doesn't work, I'll look at hacking the Bonnet like you linked, although hopefully won't need to do that since this has worked before without the mod.

Wonder if it could just be as simple as the --led-row-addr-type, guess we'll find out. Worse comes to worse I can always buy an Active3 Adapter as well.

I'll mess with it some more tonight and get back to you 😃

RoastedPecans, 2020-04-09:
So, after messing with everything again last night still made no progress.

I tried all the options available for --led-multiplexing and for --led-row-addr-type with no success. I also tried setting the panel type to the two supported types I could find in the documentation.

As previously mentioned, we have gotten this to work before with pin 8 on the Bonnet soldered for an E-Line. I can't find any specific information on my panel from the manufacturer, but I have emailed them asking for more specifications. For now, I'm assuming our panels have an E-Line since this set-up has worked with that same Bonnet pin soldered before. 

For a panel with an E-line, the led-row-addr-type should just be the default of 0 correct?

I'm starting to think that maybe there is an issue with our Bonnet. Does anyone know of anyway I could test it? Seems like getting an Active3 adapter might be the easiest way to solve my problems at this point.

hzeller, 2020-04-10:
did you also try the `--led-panel-type` ?

hzeller, 2020-05-05:
Haven't heard back, did you resolve the issue ?

RoastedPecans, 2020-05-08:
Sorry for the lack of response, got caught up in everything going on.

We did also try the `--led-panel-type` options but again with no success. At this point I think we will just purchase an Active3 adapter and see if that resolves our issues.

Thanks for the follow-up and assistance!

closed 2020-05-08 by RoastedPecans