hzeller/rpi-rgb-led-matrix #1045: Discussion: expose argument parsing?

gingters, 2020-04-10:
Hi @hzeller,

I'd like to discuss exposing the logic that does the argument parsing through the C api for my C# binding. Reason being I am currently in the process of porting all the examples to C#, and at the moment I don't have all information available.

For example, if I start my sample with `--led-chain=2`, then the library does parse this, but the changes are not reflected back into the options (which get copied over to the struct). The scrolling text example needs the chain length information and the C++ version gets that from the updated matrix_options after parsing.

I did some experiments (as I already mentioned somewhere else I am really no C/C++ dev, but I seem to find the correct spots), and I like to propose the following changes:

1. I would add the RuntimeOptions struct to the C api  
  then we can set the gpio slowdown, daemon, drop privileges and init gpio also from C# code without the need to pass them along as injected command line args
2. I would like to expose the ParseOptionsFromFlags() method, via wrapper that takes the arguments and the two structs (options & runtime options)

For that to work, in my current draft, I re-used some of the code that copies the C to the C++ struct and back and copied the method for the new runtime options struct.

Then in the binding code I can create an options object which maps the properties to the two interop structs, and call the parse method from there.
The library will parse the command line, update the structs, and so my C# options object can read the updated values back. So my code knows what settings are set in the library, without the need to manually duplicate the parsing code in C#.

Would you, in general, be willing to accept such a PR?
With further discussion about implementation details, of course. When I put the struct-copy code into a separate function I got a warning about a unused argument, and couldn't figure out why, so I disabled that warning 🙈. Maybe you would be willing to give some hints on how to improve that, too?

hzeller, 2020-05-02:
(sorry, I was busy the last couple of weeks with some technical Covid19 responses, will try to get back to you asap)

gingters, 2020-05-08:
Absolutely no problem. No hurry here. ;)
Real live always goes first and hobbies are a bit of lower priority.

TDHolmes, 2020-09-06:
I'm also interested in having the `RuntimeOptions` struct added to the C API for my rust bindings

commit 55fa32fc (2020-09-07): Expose runtime options to C API (#1141)

* Expose runtime options to C API

In other language bindings (rust in my case), I'd like to fully control
how the matrix is initialized without relying on argc/argv, as rust
doesn't even give an easy way to access this C-like representation of
command line input. Rather, I'd like to do my own command line parsing
and fully control the options.

Breakdown of changes:
- add RGBLedRuntimeOptions struct to C header
- add `static_assert`'s to make sure C/C++ structures are at least the
same size
- Adds missing `do_gpio_init` initializer to C++ `RuntimeOptio [... cut]
closed 2020-09-07 by hzeller