hzeller/rpi-rgb-led-matrix #1047: Provide a way to turn the matrix on/off

amantalion, 2020-04-11:
Hi,

Please could you tell me shat is the proper way to power the matrix temporarily off? I found this code in the RGBMatrix for doing that:

```
if (updater_) {
    updater_->Stop();
    updater_->WaitStopped();
}
delete updater_;

// Make sure LEDs are off.
active_->Clear();
if (io_) active_->framebuffer()->DumpToMatrix(io_, 0);
```

Then, to power the matrix back on I would use:

```
matrix->StartRefresh();
matrix->SwapOnVSync(&frameCanvas);
```

Is that correct? If so, then shall I create a pull request to add the function for powering the matrix off?

Thank you!

hzeller, 2020-05-05:
Is there a particular reason you'd like to do that ?

jeremywww, 2021-10-01:
To stop the CPU usage when the display would otherwise be blank for an extended period of time.