hzeller/rpi-rgb-led-matrix #1049: Quick question about wiring

salazarhog, 2020-04-13:
I went through the wiring documentation and it mentions that we have the freedom to modify the pin out.

Was there any particular reason you chose the current layout? Any pins that must be assigned to the particular GPIO to get optimum output? 

Or my can modify it any way we want without any quality issues?

Why aren't GPIO 0,1 used?

salazarhog, 2020-04-13:
I have tried switching between GPIO 2,3 with 0,1 and works perfectly
So just wondering why they weren't used. Thanks!

gingters, 2020-04-14:
I can't say anything about why Henner chose that layout, what I can say, however, is that pins 27 & 28 are ID_SD and ID_SC for the EEPROM I2C Bus and both aren't GPIO addressable. It's not that you should not use them, it's more that you really can't. They simply don't have an GPIO address and you can't control them by code.

Besides these 2 pins, I see that Pin 3 and 5 (SDA1 and SCL1 for the other programmable I2C bus) are reserved for the third chain, so if you only use one or two chains, you possibly could use the other I2C bus as these pins would be free.

I also see that the Pins 29 to 40 are reserved for chains two and three. So I _guess_ that if you only use a single chain, you have these 9 GPIO pins free at one side of the connector to connect other stuff and to not cross wires with the first chain.

gingters, 2020-04-14:
Addition:

If you only have 2 chains and left the third out, and you don't need the E line (which you only need for 64row panels, _if I understood that correctly_, you probably could also use pins 8 and 10 for UART.

salazarhog, 2020-04-14:
As I mentioned I have already successfully used GPIO 0 and 1 with the library so you're comment that we can't control them by code is plainly wrong. 

Besides, I don't get the point of your answer. I never asked which pins are free and which are not. I just want to know the reason for the current layout. Thanks anyway!

hzeller, 2020-04-14:
The Layout was chosen to 

  * be able to use a single chain with the Pi1 (which was still a thing at the time). So the first chain only uses GPIO pins available in both.
  * tries to keep the more important things such as SPI bus and serial interface last; so if you only use chain 1 and 2, there is still relevant stuff free.
  * GPIO 0 and 1 shall not be used, so they are not used by default. They do work.

This is why the pin-out looks somewhat wild, but there is a reason behind it. You can choose to use whatever pin-out you want as long as the OE pin is on 18 or 12.

salazarhog, 2020-04-14:
Thanks for the input. So as long as I don't use an EEPROM header it will be safe to use GPIO 0 and 1 in theory.

arahasya, 2020-04-20:
According to the official website
https://www.raspberrypi.org/documentation/usage/gpio/

Hardware PWM available on GPIO12, GPIO13, GPIO18, GPIO19
 So we can use OE on these 4 pins right?

closed 2020-04-20 by salazarhog