hzeller/rpi-rgb-led-matrix #1051: making one 64x64 out of two 64x32

gitmohr, 2020-04-13:
Hi!

Can somebody tell me (rookie) what the best (easiest) way is to use this code with two 64x32 (P5-16S) to make it appear as one 64x64 display?

Is it using 2 parallel chains? (I have an Adafruit HAT that only supports one chain...)
Or modifying the code to transform the pixels for the second display? How / where?
Or something else?

I am grateful for all hints, opinions  and suggestions.

Thanks!
Alex

hzeller, 2020-04-13:
Look at the pixel mappers. Simplest in your case would probably be to use the U-mapper

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#u-mapper-u-shape-connection

with that, you only need one output (like the Adafruit HAT)

gitmohr, 2020-04-13:
Indeed! Working like a charm.
THANK YOU very much, Alex

closed 2020-04-13 by gitmohr