hzeller/rpi-rgb-led-matrix #1052: Is it possible to make a matrix of 1:4 scan to work as 1:8?

salazarhog, 2020-04-14:
I have many 16x32 1:4 scan displays.

By 1:4 it means that 4 rows will be turned on.
Is it possible to imitate the behaviour of 1:4 to 1:8 with the controller?

We will have to turn on 2 rows only from a selection area of 8 rows.
Can you point me where exactly do I need to make this change in the frame buffer?

Purpose: To visualise the whole display with 1:8 scan performance which are usually cheaper.

hzeller, 2020-05-05:
I don't quite understand what you want to do, but 1:4 scan or 1:8 scan is a property of the LED panel that you can't easily change.

What you can change in the rpi-rgb-led-matrix is the choice of `--led-multiplexing`, which is the option to switch between panels of different multiplex-size.

daveythacher, 2020-07-05:
The software rewrite would be a little painful. It would reduce overhead of having to transpose one display type onto another. The underlying operation is basically a memory map and a single GPIO register with up to three data channels. Using this with address multiplexing reductions you can make this work.

If the multiplexing reduction is not supported you can make double buffering or draw two frames. Note if you do not want to change the image it may need to stay the same for both. Performance implications can exist here, power consumption and CPU usage.

Note this will reduce the number of parallel chains you can have from three down to one. To avoid this you can use FPGA to convert the address lines into two channels. Or reclaim the three back. Here the FGPA handles the conversion for you. Basically the underlying operation is a demultiplexer with state machine. However depending on implementation this may be application specific since it will likely need to know the number of clock pulses or column depth.

This same flexibility can exist in FPGA controller. The single GPIO bank approach does have some draws but this same property can be installed FPGA. Basically the software rewrite just defines a struct for that bank, hence why this rewrite could be painful. The software may not have support for modularizing the data side of the connectors. It supports the addressing which is more common to vary.

You can also force the double buffering to not work using the output enable signal and that third address line. This seems stupid to me, but it is doable if willing to waste it. Also this assumes that you do not want to deal with double buffering and cannot reduce multiplexing down to two address lines.

I tend to be a fan of hardware approaches for IO, since you can just remap it on the fly. Software can do this, but I am still critical of the underlying software notion here. It is do able, but there is an issue for certain applications. Since this project presents itself as library without disclosure of this premise it can be missed.

We are really talking about arrays here, which are compile time. These are cost optimized. They are also FPGA optimized. The issue is they can be MCU/MPU friendly with correct controller. I have been critical of a few interfaces in LED matrixes because they fail to factor in certain things. 

I see these panels speak to generations, with each generation solving an issue. The first few used LED drivers in mass scale driving down price. Using simple interfaces to avoid performance loss and cost. Next came IO reductions for cable cost. Next came bandwidth increases for density increases via PWM. Later using memory maps to improve bandwidth via locality.

Many of these use remote controller, which will need to change for improved density and application flexibility. Currently there is a few issues in control logic for MCU/MPU that is not an issue for FPGA, could be solved with silicon change. (State machine variable.) There is also an issue for the dual buffer, which does not support remote copying. These could be fixed internally or via local controller which drives several chips.

This would allow using MCU/MPU and not require huge amounts of memory. I am not sure if the need to read from the buffer exists. If it does it could exist in controller not driver logic. The MCU/MPU also would not really need to work with timing issues if it uses standard interface, which would allow it to offload via DMA. (Assuming no memory/IO/system bus limitations exist, which would not be expected with appropriate processor selection.)

Thus allowing the application to be as real time as required for frame generation. Linux could be used for compression/streaming as an example. Currently many of the controllers are custom which is probably better for security unless this is what the use internally. However many of those controllers work under fixed voltages, but this has improved with the memory mapped PWM drivers which have software gain controls.

The more things are repeated the more likely it can/should be done in hardware. Software is used for varying/glue logic. Software sucks at IO. Hardware is like a software library which abstracts logic. Software can describe hardware computation in parallel and serial. (Depending on notion of software.) Software will always be more power consuming. Hardware can cost more, but it will always be the fastest.

daveythacher, 2020-07-05:
It would be helpful to mention pin configuration. There is a couple configs that could exist here.

daveythacher, 2020-09-27:
There are two reasons to do this:
1) Lower power consumption
2) Increase Chains/Panel Size

There are three ways I can come up with:
1) Rewire them to be long 1:8 panels.
2) Multiplex the 4 data sets into 2 data sets
3) Double buffer to average the refresh rate into 1:8

1 and 3 lower power.
1 and 2 lower pin count which allows more chains.

Likely you will end up with a hardware change for the pin count. However a software rework is likely true for all. The double buffering is possibly the one you could get away with in software. However I think you are asking a lot from this type of software.

It is a large topic.

closed 2025-11-27 by marcmerlin