hzeller/rpi-rgb-led-matrix #1056: OverflowError when using SetImage inside Python thread

Kleinspider, 2020-04-17:
Thanks for making this great project! Everything has been working great except I cannot seem to get the SetImage function to work from inside a thread in Python. The same code works when it is not inside a thread, but whenever I call SetImage from inside my "renderer" thread I get an OverflowError. 

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/led-matrix-controller/matrix/renderers/main_renderer.py", line 87, in run
    self.current.render(self.canvas)
  File "/home/pi/led-matrix-controller/matrix/renderers/weather_renderer.py", line 55, in render
    self.__render_forecast()
  File "/home/pi/led-matrix-controller/matrix/renderers/weather_renderer.py", line 62, in __render_forecast
    self.matrix.SetImage(im.convert('RGB'), x, 13)
  File "core.pyx", line 22, in rgbmatrix.core.Canvas.SetImage
  File "core.pyx", line 46, in rgbmatrix.core.Canvas.SetPixelsPillow
OverflowError: can't convert negative value to size_t
```
I saw there were a couple other Issues that mentioned this error but I am not sure if there was ever a solution/workaround given for this Issue. Thanks!

hzeller, 2020-05-05:
Sounds like something happening internally to the Python mapper where some negative value is converted to an unsigned value, which you have to dig into.

cagedev, 2020-07-17:
I ran into the same issue. A look at [bindings/python/rgbmatrix/core.pyx](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/rgbmatrix/core.pyx) shows `SetImage` has a default `unsafe` mode. Calling it with `unsafe=False` worked for me.

The comments here also refer to unsafe using 'unsafe pointer acces':
#In unsafe mode we directly access the underlying PIL image array
#in cython, which is considered unsafe pointer accecss,
#however it's super fast and seems to work fine
#https://groups.google.com/forum/#!topic/cython-users/Dc1ft5W6KM4

cagedev, 2020-07-17:
#795