hzeller/rpi-rgb-led-matrix #1057: change the CLK connected to GPIO17

jhoonpp, 2020-04-22:
I want to change the CLK connected to GPIO17. I want to lower it to kHz, but is there a way?

hzeller, 2020-05-05:
The clock is generated by wiggling the GPIO pin. If you want to drastically reduce that (but why?), you'll have to add a `usleep()` at that place.

closed 2020-05-05 by hzeller

jhoonpp, 2020-05-07:
Where should you add usleep? (Because a high clock is not required and the clock is high, EMI is generated and the clock is lowered)

hzeller, 2020-05-08:
You need the high clock rate, as the pixels have to be clocked in at that rate for a stable image.
You need a few Mhz for this to work, kHz would take forever.

Each update of a single 32x32 panel needs `32 horizontal pixels * 11 bitplanes * 16 double rows = 5632' clock cycles. Then you need at least 100Hz refresh rate, so that means at least 563200 clock cycles just to clock in the data. And between these the LEDs need to be switched on the according time per bitplane. So you will not get a usable output if you'd lower the clock frequency. This has to be repeated continuously as the LED matrix does not have a frame-buffer.

If you have EMI issues, you need to follow the necessary steps for that: short cables and shielding; possibly impedance matching of output and input of your digital lines.