hzeller/rpi-rgb-led-matrix #1059: RPI4 + JOY-IT work fine for 8 sec then go crazy

StatBBS, 2020-04-22:
I have a raspberry pi 4 with a JOY-IT LED-Matrix Controller Hat
my lcd matrix is 2* 64x32 P2.5
i use a webpage that i have created to show image on my lcd matrix

The command is : sudo led-image-viewer -f -w 3 ../temp1.gif ../temp2.gif ../temp3.gif ../temp4.gif --led-rows=32 --led-cols=64 --led-chain=2 --led-brightness=50 --led-gpio-mapping=regular --led-rgb-sequence=RBG --led-slowdown-gpio=2 --led-daemon

I have tried with --led-slowdown-gpio=(1,2,3,4)
but it still dosent work 

https://drive.google.com/file/d/151oRx9ff1OCPpu1kTf10tKb6WBuLlfYs/view?usp=sharing

StatBBS, 2020-04-22:
Ok i am dumb
I have forget to do:

cat <<EOF | sudo tee /etc/modprobe.d/blacklist-rgb-matrix.conf
blacklist snd_bcm2835
EOF
sudo update-initramfs -u
sudo reboot

closed 2020-04-22 by StatBBS

hzeller, 2020-04-22:
if it goes crazy after a few seconds, this can also be an indication that the one-wire interface is not disabled.