hzeller/rpi-rgb-led-matrix #1060: Emulator

LDuncAndroid, 2020-04-25:
Sorry to be a pain, waiting on some parts arriving from Ali Express. Will there ever be an emulator available?

arahasya, 2020-04-25:
You can try to create an image using the r, g, b values

hzeller, 2020-04-27:
So in [FlaschenTaschen](https://github.com/hzeller/flaschen-taschen/tree/master/server#terminal) I wrote a simulation layer that just outputs things on the terminal. I first wrote this in the [Terminal image viewer](https://github.com/hzeller/timg). Advantage is, that you don't need any graphical user interface, just the terminal. Which is in particular good as we are running the Raspberry Pi in headless mode and are only connecting to it via ssh.

Maybe I should break that out as library, and implement a [Canvas](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/canvas.h#L32) with it.

I am currently a little busy with other projects, but if you want to work on this, this might be a good starting point.

monache2403, 2020-05-12:
Hi,
I found that (I'm French ; sorry for my poor english !) :
https://github.com/BNNorman/rpi-rgb-led-matrix-animator/tree/master/Simulator

LDuncAndroid, 2020-06-04:
Thank you both, @hzeller and @monache2403 for your responses. I'll take a look.

karanzabal, 2020-08-14:
First of all, thank you for all the work you have done with this fantastic library! (and sorry for my poor english too!)

Allow me to add myself to L Duncan's request: I have built a sports scoreboard and programmed it in c++([this one](https://sites.google.com/view/marcador-electronic/)), but it has been a long time since I stopped working as a software engineer, and although I have tried to collaborate with this part of your library, I am unable to dig deep enough into your canvas object and your FlaschenTaschen project...
Indeed, I think it would be the best way: it would allow me to add functionalities to the scoreboard without having it directly in front of me, working with a second raspberry. I think this will happen to all those who have projects based on your library in operation.

Thank you thank you thank you! I've really enjoyed these months of programming!

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828