hzeller/rpi-rgb-led-matrix #1061: Half of panel unable to display red

bsosnader, 2020-05-04:
I am using this library with a raspberry pi 3 A+, adafruit rbg matrix bonnet, and one adafruit-sourced led panel. I'm running raspbian lite and set up the library with the latest commit.

Half of the panel is unable to display red when running the demos. With the input connector at the bottom of the panel, it is the right half that is not displaying red.

Initially that half was not displaying red _or_ green, however green began displaying due to some unknown change I made (possibly switching from raspbian to raspbian lite? Tightened connections?). Now, the only issue is red.

From the small number of similar issues I found, I see this could likely to be a wiring issue. Adafruit puts red through GPIO 5 and 12. I did not solder anything and am relatively sure the bonnet is securely connected to the pi and the panel.

What else can I try before attempting to replace the adapter or panel? Thanks!

bsosnader, 2020-05-04:
Physically pushing the adapter board down with my finger sometimes will cause green to switch off and on. So, green was not working due to loose connections.

Is there anything else at all that could cause the red issue other than this? Or is there any way to salvage the bonnet to get full functionality? 

I don't currently have access to my soldering iron to try direct wiring :/

hzeller, 2020-05-04:
If you have switched off the various things that might interfere with GPIO pins (described in the [Troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) ) and are using the correct `--led-gpio-mapping` flag, then this sounds like a faulty LED panel or Bonnet. You probably should ask the Adafruit support.

bsosnader, 2020-05-04:
Did further testing and yep seems like a faulty connection on GPIO 5, made a new mapping where r1 and g1 were swapped and same thing for green. So I guess it's the bonnet. Oh well. At least that's cheaper than a new panel. Thanks

closed 2020-05-04 by bsosnader