hzeller/rpi-rgb-led-matrix #1064: TC5020A controller

tecnopolis-ve, 2020-05-09:
Hello folks!

Thanks for this great library, I'm trying to make it work with a 64x32 coreman matrix RGB led with not success so far.

I've checked the wiring (I'm using power shifters) and the library doesn't throw any error, in fact, the screen powers on but shows only random colors.

I would like to know if this controller is supported or anybody has found a wait to make it work.

Thanks!

[image]

arahasya, 2020-05-13:
Post the panel specifications and command you are using to run the code

tecnopolis-ve, 2020-05-15:
Hello!

This is the model (bought on aliexpress)

https://www.aliexpress.com/item/33049118306.html?spm=a2g0s.9042311.0.0.52f963c0JAsdMM

Manufacturer: Coreman
Chip: 74HC245 (the nearest to hub75 connector)

I'm trying to run this comment with not luck so far:

`sudo ./demo -D0 --led-rows=32 --led-cols=64 --led-slowdown-gpio=4`

I've wired the matrix led with a mcu chip and works perfectly.

Any tip is greatly appreciated.

Thanks!

arahasya, 2020-05-15:
Try all the multiplexing options
--led-multiplexing 
Make sure you have wired properly 
Have you used the E line?

tecnopolis-ve, 2020-05-15:
Hello, I've tried all the multiplexing options with no luck so far.

I've following the wiring instruction carefuly (from the repo) and still not luck.

I'm not sure whether the display is fully supported, I have no idea why with a mcu 8266 works perfectly following this wiring model:

https://github.com/witnessmenow/LED-Matrix-Display-Examples/blob/master/Wiring.png

arahasya, 2020-05-16:
I have a TC5020A display 16x32 witb 1/8 scan which works fine

maksun, 2020-05-29:
Hello, I tested two chained 64x32 panels (1/16) with TC5020A and I manage to make it working with --led-row-addr-type=4, example:

sudo ./demo -t 3600 -D 1  --led-brightness=20 --led-row-addr-type=4 --led-chain=2 --led-parallel=1 --led-cols=64 --led-rows=32

I hope it will help.

tecnopolis-ve, 2020-05-30:
Hello! @maksun 

Thanks a lot for the tip but unfortunately didn't work with my panel. I only get random patterns in the screen.

I'll buy another panel from another provider and try again, I give up with this one.

Thanks!

jerkwagon, 2020-08-05:
I bought from the same guys, 1st bactch used a different chipset, then i got some less expensive ones and they didnt work, they look just like yours, --led-row-addr-type=4  fixed my issues,  #1111  i used @maksun code he suggested and it fixed mine. maybe your wiring is wrong?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828