hzeller/rpi-rgb-led-matrix #1065: Raspi-4 vs. Raspi-3 issue

danielcloran, 2020-05-11:


hzeller, 2020-05-11:
So you say it works fine in the Pi3 but not the Pi4 ?
The Pi4 is faster, so `--led-slowdown-gpio` might need to be higher.

How are you connecting the panels, are you using an [active3 adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) ? If so, I would also suggest to use parallel chains, e.g use `--led-parallel=3 --led-chain=2` which allows for higher refresh.

danielcloran, 2020-05-11:
Leds are connected using the red board you recommended in wiring... works perfectly normal and fine on Pi3. I used --led-slowdown-gpio=2 through 4 which helped with the flicker but did not change the lines. 

I have tried using --led-parallel=3 and --led-chain=2 but I got many issues with only half of the panels lighting up so I postponed getting that to work to a later date.

hzeller, 2020-05-11:
this is strange; did you disable all potential services that might mess with your GPIO pins (such as one wire etc. mentioned in the troubleshooting section).

I've heard a few success reports of people happier with the DietPi distribution as it is more bare-bone and contains less things that might interfere with GPIO pins.

Only every other line or double line with double space showing up is usually a problem with one of the ABCDE lines not properly connected, but if you use the same adapter board and same LED matrix and stick it on different Pis, with only one having a problem, then at least the variable (that the adapter board might be broken) is out.

Are all your grounds between LED panel and Pi properly connected ? Usually that should be taken care of the adapter board, but it can't harm to connect the ground of your supply of the Matrix with the ground of the Pi (if you are using different supplies).

danielcloran, 2020-05-11:
So I downloaded DietPi last night with a clean install as I saw you mention it elsewhere and a clean build and once again had the same two line issue. Which to me ruled out any Debian buster update that might be hurting it. 

I disabled everything mentioned in troubleshooting (this pi did not have one wire enabled)... and earlier today I retested the setup on the Pi3 and everything still worked fine.

Grounds are connected through the adapter board but I just ran a manual jumper between the two grounds and nothing changed :/ 

Notable to mention is that this is a completely clean install of Debian Buster lite without anything on the pi other than your library and the essentials needed to get and build it.

danielcloran, 2020-05-11:
I could possibly order another Pi4 just to see if I got some weird defect...

hzeller, 2020-05-11:
it does sound like some GPIO issue, yep. I once destroyed part of my GPIO pins by accidentally putting on an adapter board shifted by one pin ...

danielcloran, 2020-05-11:
Argh okay... I'll put an order in now. I'll update this when I test it with the fresh Pi.

Thanks for the help,
Daniel Cloran

mikecann, 2020-05-14:
@danielcloran I too am building a cube :) we should have a chat sometime and swap notes. As for your particular issue you may want to see this thread: https://github.com/hzeller/rpi-rgb-led-matrix/issues/989

danielcloran, 2020-05-16:
New raspberry pi 4 arrived today and fixed the problem! I assume I blew one of the GPIO pins on the other one somehow... will be more careful in the future!
Thanks for all the help @hzeller.

@mikecann I'd love to chat! I looked through your thread when trying to run the panels in parallel... but didn't get very far as my panels had some different issues. I'll email you when I get a chance tonight.

mikecann, 2020-05-16:
@danielcloran ye no worries mate, my email is: mike.cann@gmail.com hit me up :)

closed 2025-11-27 by marcmerlin