hzeller/rpi-rgb-led-matrix #1067: What is the best way to implement a kind of diffusion from a central pixel to its surrounded pixels

croisez, 2020-05-17:
I was thinking that it could be of interest to implement a diffusion of a pixel to its surrounded pixels. It could be an equivalent of an anti-aliasing process.

I tried to dig into the code, and arrived at the FrameCanvas::SetPixel() function, and tried to modify it to implement the pixel diffusion.
But I was searching for something like FrameCanvas::GetPixel(), but could not find it. 

So, how could I determine the RGB values of a pixel, given its coordinates?

hzeller, 2020-05-17:
You have to implement a canvas (Canvas is just an interface) that delegates to another Canvas but keeps the pixels in an array so that you can implement getPixel().
The FrameCanvas internally converts it into the memory representation needed to push it to the matrix and there is no cheap way converting it back.

croisez, 2020-05-18:
Can you point me to some example, I have no idea how to create such
delegated Canvas