hzeller/rpi-rgb-led-matrix #1070: Panels with ICN2053BP: won't turn on

marcmerlin, 2020-05-19:
[image]
[image]

marcmerlin, 2020-05-19:
Ok, I found this bug which is apparently the same chip, or close enough to it
https://github.com/hzeller/rpi-rgb-led-matrix/issues/466

daveythacher, 2020-05-20:
These are gen 2 chips and will likely become more common. Especially for larger panels. The way they work is by using PWM generators on the chip itself rather than forcing the remote controller to emulate it across the wire. This emulation sucks up a decent amount of bandwidth which limits chain length for high FPS panels. 

Basically they need more controllers to make larger displays without this. Controllers would probably be more costly per pixel under this model. I have not seen it but I suspect they will eventually work around to gen 3 chips which will include many of the controller function directly into the chip.

That would be the next logical bandwidth reduction step to increase display density while keeping cost down. We will see I guess. I know that is where I ended up in my little side project, but for slightly different reasons.

Note there is a couple versions of these chips, but they play mostly by the same rules. Just the number values of the counters vary. Which means a library could exist, but currently I am not sure it does. There is code targeting one of them, but I have not personally played with it. 

The code looks reasonable based on the application note. I would be curious to know the exact behavior of "dead time". It uses a lot of bit banging. This was carried over from the original which worked under different environmental factors. That's why the code struggles a little too.

Hzeller will probably not do it, but I would recommend a microcontroller based interface solution for these chips. Timing can be critical, and overhead is better served handled by microcontroller. It can clean up delays in Pi logic. I see no point violating user space like that. This is a library too.

If your going to do Linux, do it right. I do not see Linux as RTOS like FreeRTOS. Even then I would not bit bang unless really put into corner. I see Linux as non-real time stuff like drawing frames into the non active buffer. No point wasting a ton of cycles on it, granted these chips do protect you here. Its IO you hand it to kernel and let it DMA it for you while you do other things.

marcmerlin, 2020-05-20:
@daveythacher that's interesting, thank you.
On the plus side, it sounds like those chips would indeed allow bigger strings with a lot less work required to refresh them.
On the minus side, they managed, yet again, to make a new chip that is incompatible with the old one (it shouldn't have been hard to support a new mode and be backward compatible with the old bigbang uC driven PWM).
@hzeller if you're interested in seeing one of the panels I got, ping me and we can probably arrange something.

daveythacher, 2020-05-20:
Do you have a link for these? How much did they cost and what was the lead time?

marcmerlin, 2020-05-20:
@daveythacher : Sure, I got them from https://www.digitspace.com/products/modules/5050-rgb-led-module/p2mm-indoor-128x64-pixel-smd-led-screen-panel-256mm128mm
It took maybe a week with fedex.
Please Email Digitspace <market@digitspace.com> and mention my name. Ask for the same panels I got, given that I just complained saying that they were not the RGBPanels people would expect, given the new chip.

marcmerlin, 2020-06-02:
@daveythacher you know what, I'm not going to use my 4 panels. If you are in the US, shoot me an Email and we can discuss options about getting them to you: marc at merlins.org

marcmerlin, 2020-06-02:
@hzeller  same deal, if you want one fo those panels, let me know.

hzeller, 2020-06-02:
I am happy to accept pull requests, but I already have more panels than one should reasonably have and also other projects competing for my time, so I might not have cycles for it in a while...

marcmerlin, 2020-06-03:
@hzeller  no worries, I don't need it to work, I'm not going to be using those panels.

daveythacher, 2020-09-14:
@marcmerlin If you still have and are still interested in letting them go, can you make ebay buy now or make offer?

marcmerlin, 2020-09-14:
I'm interested in giving one to someone who can promise to work on a driver for them for this codebase. If you have already contributed to the code, or are confident that you can,  Email me at marc<at>merlins.org and we can discuss details.

daveythacher, 2020-09-23:
For future reference Marc Merlin's, aka a "Peacock Engineer" as he refers to himself, link above is not good for getting MM mapped LED drivers. Which the ICN2053 is. This is also why it is not supported by this project. 

I wanted this for my own project and possibly for a new library. However I got a FM6126A instead, which is completely useless to me. For the life of me I do not know why a 128x64 panel is running shift registers. I do not like wasting 40 dollars on a panel which I cannot use. My long term usage for this panel was questionable as it was.

I thought about it some and came up with an outline for a new library built around the MM drivers. I was not planning to target just Pi. In fact for the first pass would support Windows and Linux. It would still be a mess, but I reflected back on a few things I looked at for my personal project. I have been at this for years, trying all kinds of different aspects. 

I think it is do able and would be better than this project. I never moved forward cause I was mostly interested in the shift registers which require a huge bandwidth for decent FPS, frame size, and color depth. I knew MM was superior ever since I played with the HT1632C/HT16K33s back around 2011/2012. Honestly would have loved to use them, however I could never find them reliably and the ICs/custom panels cost too much.

This new library version would still be terrible, but passable. In concept it would support just about any programming language given enough underlying hardware to process the language overhead. However it would still have limits, but the limits are pushed back compared to this project.

Using standard parts you can get easily you could make it work. It should be bread boardable. It is a funny one because it cheats a lot of fundamental rules. It is one of the few examples of "its just software", "let the tool do its thing",  etc. If you pass the right things to hardware for certain displays you actually can cheat the massive unforgivable sins of bit banging/software.

This version would avoid BCM issues, but would likely still have power issues. It would have issues chain length and/or frame rates. However CPU usage should be lower. You can solve the interface speed issues. Security, power efficiency, etc. if using this as control will be questionable. It would enable custom code.

My interests in Computer Engineering were embedded systems and hardware software co-design. Aka figuring out how to balance them to solve problems.

I see no reason to integrate this driver into this project. I realized it would actually be wrong due to legacy support. It would be a rewrite. The architecture of this code base is not conductive even to shift registers. However that really does not matter for them. You could do it but I see no reason to do so. I think honestly it would be a new generation of the code base.

Given the reduced overhead penalty, I think objects could be used here. However the coding structure would need to be better. I also think portability could be achieved. This would enable extension for future drivers and make support much simpler. It is a chance for better documentation and division of user interface. Areas of concern for this current project. 

However I think this project has been abused. I think its original scope was narrower, and it has been forced into what it is. It has a few prototype looks in the code, rather than a finished or release model. I find it annoying/immoral that it appears to have attempt to fill a "need" or "requirement" with such a "cost" or "we gave them something" perspective. This project has value, but only if properly discussing its limits.

I would expect this to be a challenge based on how the MM drivers handle memory between banks for MCUs. So it really is a shame really. I think I have been rebuked for this, but time will tell. Take care!

daveythacher, 2020-09-26:
I created a new code base, mostly to be annoying. It is not complete or tested. I think creating a new code base was the correct approach. Simpler, more flexible and easier to work with. It still likely sucks in a few areas, but oh well.

hzeller, 2020-09-26:
Making new code is _not_ annoying, and very welcome by everyone. Please do, @daveythacher .
I am glad for you focusing on something of value.

Annoying is poo-pooing over other peoples projects attempting to look knowledgeable (while only understanding half of the issues) by plastering their issue tracker with walls of text. I am glad you're going to move on plastering your own issue tracker with walls of text. This certainly does not risk you getting blocked on other projects.

daveythacher, 2020-09-26:
> Making new code is _not_ annoying, and very welcome by everyone. Please do, @daveythacher .
> I am glad for you focusing on something of value.
>
> Annoying is poo-pooing over other peoples projects [...]

Like why do not commit again this. You do not even understand this whole issue tracker thing. I can promise you this, if anyone is in line for being block it is you and several people on your site. I mean issue trackers are like forums dude. How do you have x number years and not know that. This is not knew.

You know you stayed quite when people hijacked issues for unrelated things. That is explicitly mention in the GitHub guidelines. Your conduct personally is explicitly in conflict. @hzeller stop it.

marcmerlin, 2021-01-27:
related thread https://rpi-rgb-led-matrix.discourse.group/t/64x128-with-2053-help-needed/99/2

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin