hzeller/rpi-rgb-led-matrix #1071: No module named Image

mickmick54, 2020-05-24:
Hi !

I managed to get the raspberry and the matrix rgb led to work with the Hat, but since an update problem with Raspbian Jessie I reinstalled Raspbian Buster. Since then I have been trying to get my Matrix rgb to work, but after installing libgraphicsmagick ++ - dev libwebp-dev -y and installing ImageViewer via the Utils folder, it no longer works. I'm looking to load PNG images through a Python script like before and it says "No module named Image". The demos don't work either. I have tried different solutions but I can't. Maybe you could help me? Will there be updates in the near future?

xuniuer, 2020-06-11:
Image module is from PIL package, or pillow. You could try to install pillow by pip.

mickmick54, 2020-07-17:
> 
> 
> Image module is from PIL package, or pillow. You could try to install pillow by pip.
Thanks for the answer I will try but I have not the time for the moment.